ros2/rviz #3: [rviz_rendering] replace hard coded resource location with ament_index

wjwwood, 2017-08-29:
In ROS 1 rviz used `rospkg` (the `ROS_PACKAGE_PATH`) to find the `rviz` package's location to find the stuff distributed in `ogre_media` at runtime. As one example, see:

- https://github.com/ros-visualization/rviz/blob/3cddcb77316e326775e52bf9ca249262cba4bda7/src/rviz/ogre_helpers/render_system.cpp#L262

Right now in the `rviz_rendering` package we have a hard coded value to find these files, which will not work long term. See:

- https://github.com/ros2/rviz/blob/7d939817a714ae3b6b4bfa3bf08b82c50cf440be/rviz_rendering/src/rviz_rendering/render_system.cpp#L279
- https://github.com/ros2/rviz/blob/7d939817a714ae3b6b4bfa3bf08b82c50cf440be/rviz_rendering/include/rviz_rendering/resource_config.hpp#L49
- https://github.com/ros2/rviz/blob/7d939817a714ae3b6b4bfa3bf08b82c50cf440be/rviz_rendering/src/rviz_rendering/config.hpp.in#L33
- https://github.com/ros2/rviz/blob/7d939817a714ae3b6b4bfa3bf08b82c50cf440be/rviz_rendering/CMakeLists.txt#L72-L73

This should instead be replaced with finding the package's installation location using the `ament_index`. The idea behind the `ament_index` is described in detail here:

https://github.com/ament/ament_cmake/blob/master/ament_cmake_core/doc/resource_index.md

There is a cmake API, which we would use to store things in the ament index if we needed to do so:

https://github.com/ament/ament_cmake/tree/master/ament_cmake_core/cmake/index

However, if we just need the installation location of the `rviz_rendering` package, that part is already "registered" in the ament index with this call in the CMake:

https://github.com/ros2/rviz/blob/7d939817a714ae3b6b4bfa3bf08b82c50cf440be/rviz_rendering/CMakeLists.txt#L111

We can access the information with the `ament_index_cpp` package which is C++ API for querying the ament index at runtime:

https://github.com/ament/ament_index/tree/master/ament_index_cpp

Here is an example of finding a package's location at runtime using the C++ API:

https://github.com/ros2/demos/blob/8354081d55c627b60b8e1229bdcd84d6ec036068/composition/src/api_composition.cpp#L89

There is a more direct API call in the python version `ament_index_python`:

https://github.com/ros2/ros2cli/blob/5584754399cb388b479814328dbf8840483ff985/ros2pkg/ros2pkg/api/__init__.py#L17

We could add that to C++ if needed.

You can introspect what is in the ament index by just browsing the files in `<install prefix>/share/ament_index/resource_index`.

Martin-Idel, 2017-08-31:
Here is my understanding of this issue:

Currently, the only "resource" we require is the ogre_media folder containing the plugins.cfg file. This is as of now linked into the program during CMake and it's linked from the source folder. Obviously, that's not the way to go.

We now want two things:

a) we want the ogre_media resources to be available in the ament_index. To do this, we can register the resources and move the folders during CMake. I would think that ogre_media was its own resource to which every plugin could then contribute, i.e. we would have a file `share/ament_index/resource_index/ogre_media/rviz_rendering` pointing at the ogre_main folder at `share/ogre_main`

b) we want the location of the ogre_media package loaded at runtime (not via CMake). This can be accomplished using the `ament_index_cpp` package if the resources have been correctly registered. Then, whenever we call `resource_config::get_resource_directory()`, we just query ament_index for the location of the ogre_main resources.

Is this the way to go?

wjwwood, 2017-08-31:
> we want the ogre_media resources to be available in the ament_index.

No, I wouldn't think of the `ogre_media` as a resource in the ament index, though that might be something we need to do for #4.

This issue could be solved by simply getting the package's location and referencing the ogre media of this package with a relative path.

The ament index will only tell you either:

- what install prefix is a package in?
- what packages have "media for rviz"? (again this is more what #4 is about)

It does not answer:

- where are the resources relative to the install prefix?
- which resource files are being exported?

You could either hard code this information (probably the right thing to do in the case of the relative location of the resources, e.g. hard code `<install prefix>/share/rviz_rendering/ogre_media`) or you could include some custom data in the resource file, which might be the right thing if you need to communicate an explicit list of resources or something. However, the content of the file is not specified by the ament index, so it could be newline separated, comma separated, yaml, or whatever you want. The ament index API will only give you the text content of the file, not interpret it.

> we want the location of the ogre_media package loaded at runtime (not via CMake).

Yes, but as I described above, I think the solution for this issue is to get the "package" resource which tells us at runtime (not via CMake) where the `rviz_rendering` package is (e.g. might be `/home/me/my_cool_ws/install_with_suffix` or could be `/usr/local`, etc...) and then we can hard code the relative path from there, i.e. `share/rviz_rendering/ogre_media` could be it.

anhosi, 2017-09-07:
We are preparing a PR for this issue.

PR #26 (2017-09-11): Get ogre_media folder from ament_index
- move ogre_media folder in installation step in CMake
- set resource directory before building the rendering system
- get the path to set the directory from ament_index

Closes #3
closed 2017-09-11 by wjwwood