ros2/rviz #4: [rviz_rendering] implement and document replacement for media_export

wjwwood, 2017-08-29:
In rviz in ROS 1, a package called `media_export` was used in combination with the `ROS_PACKAGE_PATH` via the `rospack` C++ API to allow other packages to include resources for Ogre to be used with their rviz plugins.

That code has been commented out in the ros2 port, but we need to find a suitable alternative for ROS 2. Again, this means using `ament_index` in a similar way as described in https://github.com/ros2/rviz/issues/3, but instead of using it to find one package you would use it to find packages which provide media for Ogre by looking at a different ament index resource, something like `rviz_ogre_media_exports` or similar. For details on ament index and how it could be used to do this, see resources linked to in https://github.com/ros2/rviz/issues/3.

Martin-Idel, 2017-09-04:
Let's say we have a package `my_rviz_plugin` which wants to add a few meshes to rviz. In principle, this means we need to call `getResourceLocation(...)` from the `Ogre::ResourceGroupManager`.

The ament_index can provide us with paths to all packages providing a resource. To add our plugin to the list, our plugin will add a file `my_rviz_plugin` to `ament_index/resource_index/rviz_ogre_media_exports/`. The real resources would be located somewhere next to the ament_index directory as specified by the ament_index documentation.

Now, we still haven't loaded the resources, we just know where they are. It seems we want to do the following: The `my_rviz_plugin` file contains a description containing all information of my_rviz_plugin necessary to call `getResourceLocation(...)` and adding the resources. Rviz can then load the resources at startup time when calling `RenderSystem::setupResources()`.

This would require that the format of the file `ament_index/resource_index/rviz_ogre_media_exports/my_rviz_plugin` needs to be specified, so that the parsing functionality can be implemented. Simplest solution: The my_rviz_plugin-file contains the relative path to the resource folder. We just add every file (including subdirectories?) as a resource with location type "FileSystem" and resource group of e.g. the plugin's name. 

Is this what you have in mind?

wjwwood, 2017-09-27:
@Martin-Idel sorry for not responding, but yes this is exactly what I was thinking. We can provide a cmake macro that packages (like your `my_rviz_plugin` package) can use to "register a directory" with ogre via rviz.

Reference to the old code, commented out:

-
 https://github.com/ros2/rviz/blob/ca8f42e43ba9eaa6e15434b47906d7580da1c14a/rviz_rendering/src/rviz_rendering/render_system.cpp#L333-L361

Martin-Idel-SI, 2017-10-02:
@wjwwood No problem. 
I had a further look. Here is my suggestion for the interface we want to provide:

- The resource type is `rviz_ogre_media_exports`

There are two alternatives for the macro:
- The macro follows the interface in https://github.com/ament/ament_cmake/blob/master/ament_cmake_core/doc/resource_index.md word by word, i.e. we call:
`register_package_resource(${PROJECT_NAME} "rviz_ogre_media_exports" CONTENT "../../../${PROJECT_NAME}/ogre_media/meshes" ...)` or
- the macro is tailored towards rviz: `register_rviz_ogre_media_exports(${PROJECT_NAME} DIRECTORIES ../../../${PROJECT_NAME}/ogre_media/meshes)`

Furthermore:
- I'm inclined to let the user add a list of folders to give the user all flexibility in organising resources but simplifying loading the resource (we don't consider subdirectories)
- The macro then registers the resource by adding a file to `ament_indes` containing the relative paths to all specified folders containing media and copies the media folders to the correct location.

I'll start working on it.

PR #66 (2017-10-30): Feature/rviz media export
This PR resolves #4 in the following way:

- Resources can be added as rviz_ogre_media_exports to the ament_index
- For convenience, a simple cmake macro taking the directories to be added as resources is provided
- RViz queries the ament_index at runtime adding all resources it finds
- The user is informed, if some of the claimed resource folders are missing
closed 2017-10-30 by wjwwood