ros2/rviz #6: [rviz_rendering] implement better default scene until we have display types

wjwwood, 2017-08-29:
Right now the default scene is nothing really or not displaying correctly. It is implemented in this function which will probably go away in the future:

https://github.com/ros2/rviz/blob/7d939817a714ae3b6b4bfa3bf08b82c50cf440be/rviz_rendering/src/rviz_rendering/render_window_impl.cpp#L127-L159

We need to get that working and/or update it to have some more compelling content which we can use for debugging until we have some display types ported from rviz.

greimela, 2017-08-29:
I tried inserting the grid from rviz into the scene.
The result looks like this:
[image]

I guess this should be enough for now?

wjwwood, 2017-08-29:
Looks like we might want to do something with the lighting (not sure if the sphere is just that color or if it has no light), but otherwise, yeah it looks better.

greimela, 2017-08-30:
> not sure if the sphere is just that color or if it has no light

There is light in the scene, the sphere material just does not reflect it.
We will fix this and create a PR for the issue.

PR #24 (2017-09-08): [rviz_rendering] Add grid to default scene
Adds the grid from the original `grid_display` to the `rviz_rendering` default scene.

Closes #6.
closed 2017-09-08 by wjwwood