ros2/rviz #7: [rviz_rendering] render window is black

greimela, 2017-08-29:
When running the `rendering_example` demo and switching to TAB 2 you only get a black screen instead of a rendered image.

[image]

I think this is because auto update is disabled on the `Ogre::RenderWindow`.
Removing `window->setAutoUpdated(false)` from `render_system.cpp:446` should fix this issue.

wjwwood, 2017-08-30:
This might fix the issue, but I don't think it is the correct solution, since I think the point of that is to have Qt call the ogre `renderOneFrame()` function in its render event. I'll investigate it more.

wjwwood, 2017-08-30:
Actually you are right. The lines:

```c++
  if (window) {
    window->setActive(true);
    window->setAutoUpdated(false);
  }
```

Are left over from the old rviz code, and the new example from the Ogre wiki (using the QWindow) only sets the window to visible but does not disable auto update or set it active. I'll make a pr to fix this.

PR #9 (2017-08-30): update ogre window settings
This fixes the black render window issue described in #7.

Fixes #7
closed 2017-08-30 by wjwwood