ros2/rviz #8: [rviz_rendering] CMake build fails for CMAKE_BUILD_TYPE=Debug

greimela, 2017-08-29:
When building the `rendering_example` under Ubuntu 16.04 with `CMAKE_BUILD_TYPE=Debug` the build fails with the following error message:
`make[3]: *** No rule to make target '_ogre_overlay_static_library_debug_abs-NOTFOUND', needed by 'rendering_example'.  Stop.`

wjwwood, 2017-08-29:
You need to build the `rviz_ogre_vendor` package as debug as well. It might be that the package needs additional logic to know when to rebuild when the the build type changes. To force it you can delete the build folder for that package and run it again as debug. You might also be able to delete some stamp file to force it to rebuild without deleting the build space but I haven't tried that.

greimela, 2017-08-30:
I think this was not enought for me.
I tried it with ament, which rebuilds the ogre vendor from scratch when changing the build type, but it didn't run either...

wjwwood, 2017-08-30:
Weird, I'll look again tomorrow.

Also try again now, I did clean up the cmake code some today I think.

greimela-si, 2017-09-19:
Update:
The debug build works correctly as long as I rebuild all projects after changing the build type.
So I think we can close this issue.

closed 2017-10-03 by wjwwood