ros2/rviz #18: [rviz_rendering] The build currently fails under Windows

greimela, 2017-08-31:
When building the current HEAD using ament build, I get the following error:
```
Build FAILED.
       "C:\ros2\ros2_ws\build\rviz_ogre_vendor\rviz_ogre_vendor.sln" (default target) (1) ->
       "C:\ros2\ros2_ws\build\rviz_ogre_vendor\ogre-1.10.7.vcxproj.metaproj" (default target) (3) ->
       "C:\ros2\ros2_ws\build\rviz_ogre_vendor\ogre-1.10.7.vcxproj" (default target) (5) ->
       (CustomBuild target) ->
         CUSTOMBUILD : CMake error : The following variables are used in this project, but they are set to NOTFOUND. [C:\ros2\ros2_ws\build\rviz_ogre_vendor\ogre-1.10.7.vcxproj]
```

greimela, 2017-08-31:
I noticed that ogre searches for its dependencies in `rviz_ogre_vendor/ogredeps`, but that folder does not exist.

wjwwood, 2017-08-31:
@greimela it can build freetype. I don't know why I am not seeing this issue. It might be possible that something else on my machine installed freetype somewhere. I don't know if it can be installed from choco or not.

You could try adjusting the settings in https://github.com/ros2/rviz/blob/ros2/rviz_ogre_vendor/CMakeLists.txt to either try to get it to build freetype for you or to have it look in the correct place if that is an issue.

greimela, 2017-09-04:
I managed to adjust the `rviz_ogre_vendorConfig.cmake` so that it finds freetype and zlib correctly.
PR is being prepared.

PR #22 (2017-09-11): [rviz_ogre_vendor] Fix build on windows
- Adjust the `rviz_ogre_vendorConfig.cmake` so that it finds freetype and zlib correctly.
- Build OGRE dependencies together with OGRE

Closes #18.
closed 2017-09-11 by wjwwood