ros2/rviz #25: [rviz_rendering] Scaling issue on Ubuntu + Gnome

greimela-si, 2017-09-08:
When maximising the window the render window adjusts fine.
But after unmaximising the render window does not adjust to fit the smaller window.

[image]
(The sphere should be in the center of the window)

anhosi, 2017-09-08:
I have now the same issue when unmaximising the render window.

Setup: Ubuntu 16.10 with Fvwm 2.6.5.

In the debug output the resize event for the maximise is followed by an expose event while the expose event is not present after the resize event for the unmaximise.

> [ "10:04:39:192" ]: event->type() == "QEvent::Type::Resize" in RenderWindowImpl::resize(size_t 3832, size_t 2136)
[ "10:04:39:192" ]: event->type() == "QEvent::Type::Expose" in RenderWindow::exposeEvent(QExposeEvent *)
[ "10:04:39:230" ]: event->type() == "QEvent::Type::Timer"
[ "10:04:39:230" ]: event->type() == "QEvent::Type::UpdateRequest"
[ "10:04:45:675" ]: event->type() == "QEvent::Type::Resize" in RenderWindowImpl::resize(size_t 2030, size_t 1519)
[ "10:04:45:694" ]: event->type() == "QEvent::Type::Timer"
[ "10:04:45:694" ]: event->type() == "QEvent::Type::UpdateRequest"

greimela-si, 2017-09-19:
One thing I noticed: 
The problem disappears as soon as I add an update loop to render continuously.
So we might be able to ignore this issue.

wjwwood, 2017-10-03:
We can reopen this if we notice it long term.

closed 2017-10-03 by wjwwood