ros2/rviz #45: [rviz_rendering] check if OGRE_PLUGIN_DIR is still required

wjwwood, 2017-09-27:
Since we're using static linking, rather than dynamically loading ogre plugins, we may no longer need to set it from the build system:

- https://github.com/ros2/rviz/blob/ca8f42e43ba9eaa6e15434b47906d7580da1c14a/rviz_rendering/CMakeLists.txt#L86
- https://github.com/ros2/rviz/blob/ros2/rviz_rendering/src/rviz_rendering/resource_config.cpp#L47

This is where we statically load and avoid needing to load dynamically:

- https://github.com/ros2/rviz/blob/ca8f42e43ba9eaa6e15434b47906d7580da1c14a/rviz_rendering/src/rviz_rendering/render_system.cpp#L183-L184

Martin-Idel-SI, 2017-10-04:
We checked it by just deleting the lines and rebuilding everything. No errors and it runs on Linux at least, so it seems to be unnecessary now.

wjwwood, 2017-10-20:
> Do you want a Pull-Request deleting this?

👍 Yes, please.

PR #76 (2017-10-24): Cleanup unnecessary cmake variable OGRE_PLUGIN_DIR
Resolves #45: 
I checked that it builds and runs on Ubuntu 16.04 (x64) and Windows 10 (x64)
closed 2017-10-24 by wjwwood