ros2/rviz #46: [rviz_rendering] add glsl resources and test with / fix glsl 150

wjwwood, 2017-09-27:
See: https://github.com/ros2/rviz/blob/ca8f42e43ba9eaa6e15434b47906d7580da1c14a/rviz_rendering/src/rviz_rendering/render_system.cpp#L314

Martin-Idel-SI, 2017-10-05:
Maybe we can try to help, but could you maybe clarify the issue? On my Ubuntu with NVidia graphics card, I get OpenGL 4.5 and thus glsl >150 and there seems to be no issue. Resources seem to get added and I have no error or warning. What exactly is supposed to happen or not happen? Thank you.

wjwwood, 2017-10-06:
@Martin-Idel-SI Sure, sorry for not having more details in the original issue. I was getting issues specifically with the `// include` in the glsl scripts, because ogre would parse the comment, then try to find the resource using the `Ogre::ResourceManager` thing and that was failing.

Here's an example of one of the include comments:

https://github.com/ros2/rviz/blob/ros2/rviz/ogre_media/materials/glsl120/glsl120.program#L2-L5

I'd have to move that folder to `rviz_rendering` again and try to reenable the loading of those to see the issue again. I can try to setup a branch with that on it for you tonight.