ros2/rviz #51: port resource_retriever to ROS 2

wjwwood, 2017-09-27:
rviz uses `resource_retriever` to fetch files, and specifically uses the `package://` scheme to locate files relative to a package's install location. This is implemented in ROS 1 using `rospack` and the `ROS_PACKAGE_PATH`, in ROS 2 it will need to use the `ament_index_cpp` API.

I'll take this one on too, since it is outside of the rviz repository.

wjwwood, 2017-09-27:
I forgot to mention that a temporary port/hack of this is in the "temp" folder and it should be removed:

https://github.com/ros2/rviz/tree/ca8f42e43ba9eaa6e15434b47906d7580da1c14a/rviz_common/src/rviz_common/temp/resource_retriever

greimela-si, 2017-12-05:
Has this been done in the context of #123?

wjwwood, 2017-12-05:
Yes!

closed 2017-12-05 by wjwwood