ros2/rviz #53: [rviz_common] remove catch all try-catch clause

wjwwood, 2017-09-27:
In favor of exact exceptions to catch... See:

- https://github.com/ros2/rviz/blob/ca8f42e43ba9eaa6e15434b47906d7580da1c14a/rviz_common/src/rviz_common/frame_manager.cpp#L96-L99

Martin-Idel-SI, 2017-10-19:
Judging the code as-is, an error could only be thrown while constructing the Time. We use the explict constructor here 
https://github.com/ros2/rclcpp/blob/989084b3de84000a15fb4bd5a7d63929001cd769/rclcpp/src/rclcpp/time.cpp#L98-L103

Looking at the lines above, errors can only happen when initialising time source or time point. Since we use default parameters everywhere, this shouldn't happen and if it does, it will also happen in `rclcpp::Time::now()`: 

- https://github.com/ros2/rcl/blob/38d3725262e25d5beb4c2b5c1155b74402d371d1/rcl/include/rcl/time.h#L112-L121
- https://github.com/ros2/rcl/blob/38d3725262e25d5beb4c2b5c1155b74402d371d1/rcl/include/rcl/time.h#L252-L259
and here is the code for `rclcpp::Time::now()`, which does a little bit of checking on top of calling a constructor:
- https://github.com/ros2/rclcpp/blob/989084b3de84000a15fb4bd5a7d63929001cd769/rclcpp/src/rclcpp/time.cpp#L63-L84

This implies that we cannot really implement a good fallback in case of failure, hence I believe we should not even try to catch anything at the moment.

wjwwood, 2017-10-20:
> This implies that we cannot really implement a good fallback in case of failure, hence I believe we should not even try to catch anything at the moment.

:+1: for a pr to remove it.

PR #105 (2017-10-24): Remove try-catch block with unknown errors
Resolves #53 
I removed the try catch block since:

- it uses only default ROS2 behaviour (specific errors are thrown)
- if it fails, ROS2 fails and no valid fallback can be provided (the function now() uses the same calls and would fail, too)
- hence we shouldn't catch the error at this point
closed 2017-10-24 by wjwwood