ros2/rviz #56: [rviz_common] fix loading of icon resources

wjwwood, 2017-09-27:
See:

- https://github.com/ros2/rviz/blob/ca8f42e43ba9eaa6e15434b47906d7580da1c14a/rviz_common/src/rviz_common/load_resource.cpp#L45-L62
- https://github.com/ros2/rviz/blob/ca8f42e43ba9eaa6e15434b47906d7580da1c14a/rviz_common/src/rviz_common/load_resource.cpp#L69-L91
- https://github.com/ros2/rviz/blob/ca8f42e43ba9eaa6e15434b47906d7580da1c14a/rviz_common/src/rviz_common/load_resource.cpp#L104-L118
- https://github.com/ros2/rviz/blob/ca8f42e43ba9eaa6e15434b47906d7580da1c14a/rviz_common/src/rviz_common/load_resource.cpp#L123-L165

This will probably be solved with a combination of either `ament_index_cpp` (to have a relocatable solution to finding the files at runtime) and `QDir`/`QFile`/`QFileInfo` to replace `boost::filesystem`.

greimela-si, 2017-09-28:
Will this use the `resource_retriever` from #51 or should `ament_index_cpp` be used directly?

wjwwood, 2017-09-28:
> Will this use the resource_retriever from #51 or should ament_index_cpp be used directly?

We should use `resource_retriever` if possible. If `resource_retriever` gives us a `file://` back from a `package://` then we can use that directly. Otherwise we'll have to let `resource_retriever` open the file and see if `QPixmap` has a version of `.load()` that takes a buffer/file-like object instead of a path.

wjwwood, 2017-12-05:
This should be reevaluated after #123 was merged.

PR #235 (2018-03-31): Restore icons throughout RViz2
Fixes #56. 

The package and path for a icon are derived from the class id of the plugin. This requires some renaming of classIds in the `plugin_description.xml`.

As a consequence there is a **breaking** change: an already existing `~/.rviz2/default.rviz` will not load correctly. We fixed the crash so a user is faced with an empty rviz window and needs to add all panels and displays manually and save the config afterwards.
closed 2018-03-31 by wjwwood