ros2/rviz #64: rviz_common on ros2 branch fails to build

Martin-Idel-SI, 2017-10-09:
Using `ament build` or `ament build --cmake-args -DCMAKE_BUILD_TYPE=Debug` the current ros2 branch doesn't build on Linux (Ubuntu 16.04). The error reads:

> CMake Error in CMakeLists.txt:
>   Imported target "rviz_common::rviz_common" includes non-existent path
> 
>     "/usr/lib/include"
> 
>   in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
> 
>   * The path was deleted, renamed, or moved to another location.
> 
>   * An install or uninstall procedure did not complete successfully.
> 
>   * The installation package was faulty and references files it does not
>   provide.

and more errors of this type.

As a quick-fix, comment out the line ` ${ASSIMP_INCLUDE_DIRS}` in `CMakeLists.txt` of `rviz_common`, but we should investigate the underlying issue.

greimela-si, 2017-10-11:
@wjwwood Should we fix this temporarily with the workaround above or do you plan to directly create a permanent fix?

greimela-si, 2017-10-16:
I investigated this issue a bit further.

The problems lies in the inflexibility of assimps config script `assimp-config.cmake`.
It searches the path to its include directory like so: 

> Starting from the location of the config script (e.g. `/usr/local/lib/cmake/assimp-3.3/assimp-config.cmake`) 
> - move 4 directories upwards (e.g. to `/usr/local`) 
> - and then append `/include` to this directory.

This works in the case of this example but is obviously quite unstable.
When using the default assimp package from ubuntu, the config file is placed in `/usr/lib/x86_64-linux-gnu/cmake/assimp-3.2/assimp-config.cmake`. 
Therefore, going 4 directories upwards ends up in the wrong directory.

**Possible Solutions:**
- Build and install Assimp from source
- Provide a vendor package for Assimp

greimela-si, 2017-10-16:
@wjwwood What do you think. Should we create a vendor package for assimp?

wjwwood, 2017-10-16:
Sorry, I had planned to look into this last week, but I didn't get a chance. Since we need it to be available on Windows, and it doesn't appear to be available in Chocolatey, I'd say making a vendor package (similar to the ogre one) is a good idea. However, we should make sure to use the system provided assimp where possible (unlike the ogre vendor package). In that case we can still provide a bit of cmake logic to work around any issues with the non-existent include directory.

dirk-thomas, 2017-11-15:
I just ran into the same problem. Locally I commented out the following line:

https://github.com/ros2/rviz/blob/9649866778402433ec24ff81807fb4281ef211fc/rviz_common/CMakeLists.txt#L292

But this should probably get a fix committed to avoid users running into this.

Karsten1987, 2017-11-15:
Does it make sense to set up a (dedicated) CI job for this in order to prevent `HEAD` checkouts unable to build? Looking at the last few commits, it looks to me this happened before.

wjwwood, 2017-11-15:
> But this should probably get a fix committed to avoid users running into this.

Problem is, that if I comment out this line it happens to work on Linux if you're using assimp from apt, but maybe not on Windows or macOS or if you build assimp from source on Linux. We'll need to do an intermediate vendor package to handle the issue with the invalid exported include directory on Linux I think.

If @Martin-Idel-SI or @greimela-si (or anyone else) have time, could someone make a pr that conditionally adds this include directory, with a check something like "if Linux and if the directory does not exist, then do not add it". That will work around it in this one case for now.

Later the vendor package will be needed to handle it for all cases, to avoid having to copy-paste a similar hack around everywhere we use assimp directly, and so we can build it from source on Windows automatically. So we should leave this issue open until we get there.

> Does it make sense to set up a (dedicated) CI job for this in order to prevent HEAD checkouts unable to build? Looking at the last few commits, it looks to me this happened before.

I will eventually require ci.ros2.org passing for new pr's to this repo, but I've been waiting until we get the `pluginlib` stuff ready because we'll have to add several things to the CI's repo list at that point. It's a good idea though.

anhosi, 2017-11-16:
For me this is a bug in assimp or at least the assimp Ubuntu package. Adding a workaround/ vendor package to rviz feels wrong to me. Also that would not scale very well.

greimela-si, 2017-11-16:
We created an issue in the assimp repo.
https://github.com/assimp/assimp/issues/1571

wjwwood, 2017-11-16:
@anhosi sorry, I might not have been clear. I think we do need a vendor package so that we can satisfy the dependency on Windows (at least temporarily) because there is no good way to automatically get it as a dependency, i.e. it is not in chocolatey.

Since we need the vendor package for Windows (or at least until we solve that issue another way) we could use it to fix this issue in one place. Notifying the upstream of the issue in parallel is fine, but doesn't eliminate the need for the vendor package, imo.

Also, you'd probably need to notify the Debian/Ubuntu maintainers additionally. Even if the assimp maintainers fix it, we can't utilize that fix without building it ourselves, and that, imo, implies again that we need a vendor package.

anhosi, 2017-11-16:
Right, I forgot about the missing installation package on Windows. I completely agree.

wjwwood, 2017-12-05:
There's now an `rviz_assimp_vendor` package which works around this by not using the assimp from `apt`, at least for now.

Fixed in #123.

closed 2017-12-05 by wjwwood