ros2/rviz #75: Second Window on Mac

Martin-Idel-SI, 2017-10-24:
Currently, on Mac rendering happens in a second window instead of just one as it should. 

The problem is the external window handle. When you try to use the version working on Windows 10 (and also advertised in the official wiki: http://wiki.ogre3d.org/tiki-index.php?page=Integrating+Ogre+into+QT5 ), RViz segfaults on OS X. Debugging the issue, we saw that it fails while asserting that the externalWindowHandle specifies a valid NSView.

Having a look at Gazebo, which also uses Qt 5 and Ogre3D, older code contained a comment
`// Mac and Windows *must* use externalWindow handle.`
(see https://bitbucket.org/osrf/gazebo/pull-requests/2764/gui-plugins-on-macos/diff)
Current code still needs externalWindowHandle, however they calculate it specifically, but need X11 on OSX for it:
See https://github.com/arpg/Gazebo/blob/619218c0bb3dc8878b6c4dc2fddf3f7ec1d85497/gazebo/rendering/WindowManager.cc#L84
for the externalWindowHandle on Mac, which is calculated here: 
https://github.com/thomas-moulard/gazebo-deb/blob/456da84cfb7b0bdac53241f6c4e86ffe1becfa7d/gazebo/gui/GLWidget.cc#L714-L735
So it's definitely different from the window handle in our code.

We'll investigate the issue further and post whatever useful information we find.

wjwwood, 2017-11-17:
Any progress on this one? If not, no worries, it's just starting to bug me 🙂

Martin-Idel-SI, 2017-11-20:
Sadly not. We don't have a really deep knowledge of Mac so we left it for the moment. 

Maybe we could adapt the solution Gazebo used, but that would require introducing X11-dependencies and we didn't want to try this too hard just now.

wjwwood, 2017-12-08:
This is fixed in https://github.com/ros2/rviz/pull/137

closed 2017-12-08 by wjwwood