ros2/rviz #108: rviz_rendering build fail cause rviz_ogre_vendor::OgreMain-NOTFOUND

yechun1, 2017-11-13:
I try to build rviz 2 but fail while build rviz_rendering, rviz_ogre_vendor::OgreMain-NOTFOUND. Does any one meet this issue

# build log
root@robot2:~/ros2_ws/rviz_ws/rviz2# ament build
+++ Building 'rviz_rendering'
==> '. /root/ros2_ws/rviz_ws/rviz2/build/rviz_rendering/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/root/ros2_ws/rviz_ws/rviz2/build/rviz_rendering'
==> '. /root/ros2_ws/rviz_ws/rviz2/build/rviz_rendering/cmake__build.sh && /usr/bin/make -j8 -l8' in '/root/ros2_ws/rviz_ws/rviz2/build/rviz_rendering'
[  8%] Automatic moc for target rviz_rendering
[  8%] Automatic moc for target rendering_example
[  8%] Built target rendering_example_automoc
[  8%] Built target rviz_rendering_automoc
[ 86%] Built target rviz_rendering
CMakeFiles/rendering_example.dir/build.make:123: *** target pattern contains no '%'.  Stop.
CMakeFiles/Makefile2:137: recipe for target 'CMakeFiles/rendering_example.dir/all' failed
make[1]: *** [CMakeFiles/rendering_example.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

# check the build.make 
find several modules "NOTFOUND"
cat build/rviz_rendering/CMakeFiles/rendering_example.dir/build.make
rendering_example: rviz_ogre_vendor::OgreMain-NOTFOUND
rendering_example: rviz_ogre_vendor::OgreOverlay-NOTFOUND
rendering_example: rviz_ogre_vendor::RenderSystem_GL-NOTFOUND
rendering_example: rviz_ogre_vendor::OgreGLSupport-NOTFOUND

greimela-si, 2017-11-13:
This error seems to occur when the CMake build type is not the same for all packages.

Can you try the following steps?
- Delete the `build` and `install` folder of your rviz workspace
- Build and explicitly provide the build type: 
  `ament build --cmake-args -DCMAKE_BUILD_TYPE=<Debug|Release>`

yechun1, 2017-11-14:
`ament build --cmake-args -DCMAKE_BUILD_TYPE=<Debug|Release>`, could pass this build. 

More thinking about this issue, 
`"ament build" ` will build fail, seems it use `"-DDCMAKE_BUILD_TYPE=RelWithDebInfo" `by default.

Check with "rviz_ogre_vendor/rviz_ogre_vendorConfig.cmake.in", already set_property IMPORTED_LOCATION_RELEASE|DEBUG, but not be available (rviz_ogre_vendor::OgreMain-NOTFOUND)
After set_property(IMPORTED_CONFIGURATIONS RELEASE|DEBUG), cmake could get available imported configuration. 

Do not know why Release|Debug works, but RelWithDebInfo not.  Set IMPORTED_CONFIGURATIONS may is workaround.

yechun1, 2017-11-15:
Root Cause: 
Not `set_target_properties `for `IMPORTED_LOCATION_NOCONFIG` in `rviz_ogre_vendorConfig.cmake.in`, so that cmake could not find the library. 

Solution:
-  First delete the `build `and `install `folder, then build with `ament build --cmake-args -DCMAKE_BUILD_TYPE=<Debug|Release>`

-  Or set `IMPORTED_CONFIGURATION ` and just apply the patch #109 , build with "ament build" could work.

yechun1, 2017-11-16:
Patch  #109 merged, close this issue.

closed 2017-11-16 by yechun1