ros2/rviz #126: Finish refactoring and unit testing of PointCloud

Martin-Idel-SI, 2017-11-28:
PointCloud has now been ported, however, we didn't get to do everything we want. Here is what would be necessary or nice to have but which is currently not possible:

- [ ] Introduce pluginlib (requires https://github.com/ros2/rviz/issues/49, already on branch)
- [ ] Add tests to PointCloudCommon (requires https://github.com/ros2/rviz/issues/125) + refactoring of `update`
- [ ] Add tests to PointCloudSelectionHandler + refactoring (requires a SelectionManager and therefore https://github.com/ros2/rviz/issues/125)
- [ ] Reintroduce RosTopicProperty in MessageFilterDisplay to see available topics (was replaced by StringProperty)
- [ ] Reintroduce/find alternative to tf2 message filter (not ported, yet)

greimela-si, 2017-12-06:
I extracted the pluginlib integration into a separate issue, namely #135.

Martin-Idel-SI, 2018-02-07:
Using the IntensityPCTransformer, the default visibility of some properties is not correct although it seems to be correctly set (autocompute intensity property is checked, but the manual min and max values are still visible, a similar problem is seen with the rainbow property). 

This could be investigated when introducing pluginlib to transformers.

closed 2018-06-16 by Martin-Idel-SI