ros2/rviz #132: Issues in the RViz release candidate

greimela-si, 2017-12-05:
## GUI

**Menu**
- [ ] The link to rviz wiki redirects to the ros1 rviz wiki

### Panels
**Displays panel**
- [ ] An existing error message is not removed when the error is resolved
    - e.g. the message that tf is not able to transform between two frames does not disappear after this transformation is provided
- [ ] The error message field can be edited

**View controller panel**
- [ ] The orbit view controller has stereo enabled per default. This results in the image being slightly shifted
- [ ] New view controllers can be added, but not renamed or removed

### Displays

**PointCloud**
- [ ] Decay time > 0 results in flickering points

**Grid**
- [ ] After adding an extra Grid display, it cannot be removed

## Console output
- [ ] The default scene is still loaded in the background
```
[ERROR] [rviz2]: WARNING: rviz_sphere.mesh is an older format ([MeshSerializer_v1.8]); you should upgrade it as soon as possible using the OgreMeshUpgrade tool
```
- [ ] Missing material `BaseWhiteNoLighting`
```
[ERROR] [rviz2]: Can't assign material BaseWhiteNoLighting to SubEntity of Shape1 because this Material does not exist in group rviz_rendering. Have you forgotten to define it in a .material script?
```

wjwwood, 2017-12-07:
Since #138 has been merged, I'll close this. @greimela-si can you open issues for any of the remaining issues here?

Thanks again for testing and fixing things up guys!

closed 2017-12-07 by wjwwood

greimela-si, 2017-12-08:
I'd say the one with the error message is a won't fix, since the message is immediately overwritten most of the time.
I created #145 for the wiki link issue.