ros2/rviz #133: Ros2 Release Rviz Windows build

anhosi, 2017-12-05:
List of issues and a few solutions I ran into trying to build rviz on windows.

- `tinyxml2_vendor`: This package does almost nothing when there is already a tinyxml2 installed. However, when building `rviz`the `#include "tinyxml2"` within `pluginlib` fails. I tried to remove the system tinyxml2 but then `fastrtps` will no longer build even if `tinyxml2_vendor is built first via `ament build --only tinyxml2_vendor`.
- `assimp`: assimp in version 4.0.1 does not build on windows (https://github.com/assimp/assimp/issues/1548), the current master does: changing lines

wjwwood, 2017-12-05:
W.r.t.

> `tinyxml2_vendor`

I am currently struggling with this, hopefully I'll have it sorted out soon (it is a requirement for Linux and macOS too, and it's currently causing issues).

> assimp in version 4.0.1 does not build on windows

Specifically for VS2017, we're using VS2015 so that should be ok for us. It would be nice to see a more up-to-date assimp that fixes this for VS2017 in the ROS 2 release, but it's not a critical fix for the ROS 2 release (fixing it anytime after the release is acceptable in my opinion).

> rviz_ogre_vendor: no longer builds its dependencies

That was intentional, the libraries it built had warnings and I could not suppress them, so I disabled the building of the dependencies with the aim to build them external from ogre and provide them to ogre as needed. So far on macOS and Linux this has not been an issue, but we may need to do something else for Windows (to proceed trying to fix Windows you can just enable it again).

> pluginlib does does not "build"on windows

It builds for me, that's likely another difference between VS2017 and VS2015.

> Qt: Qt is not found with the current instructions in the Readme. The CMAKE_PREFIX_PATH needs to be adjusted.

I just set the `Qt5_DIR` to something like `C:\Qt\5.9.3\msvc2015_64` (would be `msvc2017_64` for you guys).

anhosi, 2017-12-06:
At least with VS2017 I can re-enable Qt automoc after I updated cmake to 3.10.0. It seems there is a problem with cmake 3.9.
https://stackoverflow.com/questions/45764036/cmake-3-9-now-creates-autogen-projects-in-visual-studio
https://gitlab.kitware.com/cmake/cmake/issues/17205

The linking issue still remains, it seems as if only references to ogre are not resolved.

anhosi, 2017-12-06:
There is a problem with yaml-cpp: https://github.com/jbeder/yaml-cpp/issues/461. It seems that the shared library build is unusable on windows.

anhosi, 2017-12-06:
With the following changes I avoid all windows linker problems except those caused by the above mentioned yaml-cpp bug.
```diff --git a/rviz2/CMakeLists.txt b/rviz2/CMakeLists.txt
index 9621b310..4fc7a8fb 100644
--- a/rviz2/CMakeLists.txt
+++ b/rviz2/CMakeLists.txt
@@ -30,6 +30,8 @@ find_package(rviz_common REQUIRED)

 find_package(rviz_ogre_vendor REQUIRED)

+find_package(rviz_yaml_cpp_vendor REQUIRED)
+
 find_package(Qt5 REQUIRED COMPONENTS Widgets)

 add_executable(${PROJECT_NAME}_app
@@ -37,6 +39,7 @@ add_executable(${PROJECT_NAME}_app
 )
 target_link_libraries(${PROJECT_NAME}_app
   rviz_common::rviz_common
+  ${rviz_yaml_cpp_vendor_LIBRARIES}
   Qt5::Widgets
 )
```
```diff --git a/rviz_rendering/CMakeLists.txt b/rviz_rendering/CMakeLists.txt
index 826d853f..e25a7458 100644
--- a/rviz_rendering/CMakeLists.txt
+++ b/rviz_rendering/CMakeLists.txt
@@ -82,10 +82,10 @@ add_library(rviz_rendering SHARED
   src/rviz_rendering/point_cloud_renderable.cpp
 )
 target_link_libraries(rviz_rendering
-  PRIVATE rviz_ogre_vendor::OgreMain
-  PRIVATE rviz_ogre_vendor::OgreOverlay
-  PRIVATE rviz_ogre_vendor::RenderSystem_GL
-  PRIVATE rviz_ogre_vendor::OgreGLSupport
+  PUBLIC rviz_ogre_vendor::OgreMain
+  PUBLIC rviz_ogre_vendor::OgreOverlay
+  PUBLIC rviz_ogre_vendor::RenderSystem_GL
+  PUBLIC rviz_ogre_vendor::OgreGLSupport
   Qt5::Widgets
   ament_index_cpp::ament_index_cpp
 )
```

```diff --git a/rviz_yaml_cpp_vendor/rviz_yaml_cpp_vendorConfig.cmake.in b/rviz_yaml_cpp_vendor/rviz_yaml_cpp_vendorConfig.cmake.in
index af7eba57..9f8326d0 100644
--- a/rviz_yaml_cpp_vendor/rviz_yaml_cpp_vendorConfig.cmake.in
+++ b/rviz_yaml_cpp_vendor/rviz_yaml_cpp_vendorConfig.cmake.in
@@ -23,5 +23,16 @@ endif()

 find_package(yaml-cpp REQUIRED QUIET)

-set(rviz_yaml_cpp_vendor_LIBRARIES ${YAML_CPP_LIBRARIES})
+set(rviz_yaml_cpp_vendor_LIBRARIES)
+foreach(library IN LISTS YAML_CPP_LIBRARIES)
+  message(STATUS "library: ${library}")
+  if(IS_ABSOLUTE "${library}")
+    list(APPEND rviz_yaml_cpp_vendor_LIBRARIES "${library}")
+  else()
+    find_library(library_abs ${library} PATHS "${@PROJECT_NAME@_DIR}/../../../opt/rviz_yaml_cpp_vendor/lib" NO_DEFAULT_PATH)
+    list(APPEND rviz_yaml_cpp_vendor_LIBRARIES "${library_abs}")
+  endif()
+endforeach()
+
+set(rviz_yaml_cpp_vendor_LIBRARY_DIRS ${YAML_CPP_LIBRARY_DIRS})
 set(rviz_yaml_cpp_vendor_INCLUDE_DIRS ${YAML_CPP_INCLUDE_DIR})
```

At least up until the building of `rviz2`.

anhosi, 2017-12-06:
To work around the yaml-cpp bug we switched to a static build:
```diff --git a/rviz_yaml_cpp_vendor/CMakeLists.txt b/rviz_yaml_cpp_vendor/CMakeLists.txt
index 0ef8d738..f1373237 100644
--- a/rviz_yaml_cpp_vendor/CMakeLists.txt
+++ b/rviz_yaml_cpp_vendor/CMakeLists.txt
@@ -26,7 +26,7 @@ macro(build_yaml_cpp)
   list(APPEND extra_cmake_args "-DYAML_CPP_BUILD_TESTS=OFF")
   list(APPEND extra_cmake_args "-DYAML_CPP_BUILD_TOOLS=OFF")
   list(APPEND extra_cmake_args "-DYAML_CPP_BUILD_CONTRIB=OFF")
-  list(APPEND extra_cmake_args "-DBUILD_SHARED_LIBS=ON")
+  list(APPEND extra_cmake_args "-DBUILD_SHARED_LIBS=OFF")

   include(ExternalProject)
   # This specific version (past the current latest release of 0.5.3) is required to make
```

```diff --git a/rviz_yaml_cpp_vendor/rviz_yaml_cpp_vendorConfig.cmake.in b/rviz_yaml_cpp_vendor/rviz_yaml_cpp_vendorConfig.cmake.in
index af7eba57..1e967d80 100644
--- a/rviz_yaml_cpp_vendor/rviz_yaml_cpp_vendorConfig.cmake.in
+++ b/rviz_yaml_cpp_vendor/rviz_yaml_cpp_vendorConfig.cmake.in
@@ -23,5 +23,16 @@ endif()

 find_package(yaml-cpp REQUIRED QUIET)

-set(rviz_yaml_cpp_vendor_LIBRARIES ${YAML_CPP_LIBRARIES})
+set(rviz_yaml_cpp_vendor_LIBRARIES)
+foreach(library IN LISTS YAML_CPP_LIBRARIES)
+  message(STATUS "library: ${library}")
+  if(IS_ABSOLUTE "${library}")
+    list(APPEND rviz_yaml_cpp_vendor_LIBRARIES "${library}")
+  else()
+    find_library(library_abs "libyaml-cppmd" PATHS "${@PROJECT_NAME@_DIR}/../../../opt/rviz_yaml_cpp_vendor/lib" NO_DEFAULT_PATH)
+    list(APPEND rviz_yaml_cpp_vendor_LIBRARIES "${library_abs}")
+  endif()
+endforeach()
+
+set(rviz_yaml_cpp_vendor_LIBRARY_DIRS ${YAML_CPP_LIBRARY_DIRS})
 set(rviz_yaml_cpp_vendor_INCLUDE_DIRS ${YAML_CPP_INCLUDE_DIR})
```
The difference to the previous diff is the hard-coded  library name "libyaml-cppmd".

kimkulling, 2017-12-20:
There is a new assimp-version out.

closed 2018-02-12 by anhosi