ros2/rviz #140: OS X "binaries" hard-link against assimp from build machines

dhood, 2017-12-07:
Using [this packaging job for OS X using fastrtps](http://ci.ros2.org/view/packaging/job/packaging_osx/833/):

```
$ ./bin/rviz2_app
dyld: Library not loaded: /Users/osrf/jenkins/workspace/packaging_osx/ws/build/rviz_assimp_vendor/assimp_install/lib/libassimp.4.dylib
  Referenced from: /Users/dhood/Downloads/ros2-ardent-fastrtps/./bin/rviz2_app
  Reason: image not found
Trace/BPT trap: 5
```

```
$ otool -L ./bin/rviz2_app | grep osrf
	/Users/osrf/jenkins/workspace/packaging_osx/ws/build/rviz_assimp_vendor/assimp_install/lib/libassimp.4.dylib (compatibility version 4.0.0, current version 4.0.1)
```

Perhaps the equivalent of https://github.com/ros2/rmw_fastrtps/pull/128/files would apply here

wjwwood, 2017-12-07:
Yeah, @Karsten1987 and I were just seeing that too. I'll have a look into what the issue is, probably just need to set one of the RPATH settings in CMake.

wjwwood, 2017-12-08:
I think the simple answer to this one might be to install assimp on the build machines (I think that's why I didn't see this), though this issue might crop up in the debian packages where we have to build assimp from source due to needing a newer version on Linux.

wjwwood, 2017-12-08:
I've updated the installation instructions, our mac CI setup instructions, and updated the CI machines to have `assimp` installed. My hope is that this works around the issue for us, and is actually better since it makes the macOS jobs shorter, but I've narrowed down the issue and should be able to find a fix regardless, just not tonight.

dhood, 2017-12-08:
I installed assimp on my machine too and now I don't get that error with last night's packaging job so this appears to have worked :+1:

I am running into something else now that I'm looking into (might just be my machine, I'll try on another one):

```
$ rviz2_app
dyld: Symbol not found: __ZN4YAML11InvalidNodeD1Ev
  Referenced from: /Users/dhood/Downloads/ros2-ardent-rviz/bin/rviz2_app (which was built for Mac OS X 10.12)
  Expected in: /usr/local/lib/libyaml-cpp.0.5.dylib
 in /Users/dhood/Downloads/ros2-ardent-rviz/bin/rviz2_app
Trace/BPT trap: 5
```

but actually it looks like it has that symbol:
```
$ nm /usr/local/lib/libyaml-cpp.0.5.dylib | grep Invalid
0000000000002b5a t __ZN4YAML11InvalidNodeC2Ev
0000000000002be8 t __ZN4YAML11InvalidNodeD0Ev
0000000000002b1e t __ZN4YAML11InvalidNodeD1Ev
000000000001f024 T __ZN4YAML7Scanner19InvalidateSimpleKeyEv
000000000001ed98 T __ZN4YAML7Scanner9SimpleKey10InvalidateEv
00000000000300e0 D __ZTIN4YAML11InvalidNodeE
000000000002d200 S __ZTSN4YAML11InvalidNodeE
00000000000300f8 d __ZTVN4YAML11InvalidNodeE
```

dhood, 2017-12-08:
I saw that the build machines didn't have `yaml-cpp` installed from brew so I uninstalled it on my machine.

Now as long as I do `$ export DYLD_LIBRARY_PATH=/Users/dhood/Downloads/ros2-ardent-rviz/opt/rviz_yaml_cpp_vendor/lib:$DYLD_LIBRARY_PATH` I can run `rviz2_app` 😁 
@mikaelarguedas said that there's some work going on with yaml-cpp that might address that but even without that: yay!

wjwwood, 2017-12-08:
@dhood thanks for testing, we merge the pr that fixes the yaml-cpp, we need to do one more patch to fix debs and I have a fix for the library path too.

dhood, 2017-12-08:
OK. I'll close this issue since FWICT there's no more linking against absolute paths from the build machine.

wjwwood, 2017-12-08:
@dhood I've started a new "ci packaging" job for macOS here:

http://ci.ros2.org/job/ci_packaging_osx/19/

When it's done it should be possible to test out that artifact to see if the linking has been fixed.

dhood, 2017-12-08:
ok will do

mikaelarguedas, 2017-12-08:
Hopefully it should work regardless of if yamlcpp is installed or not on the target system right ?
(just to scope the testing cases)

wjwwood, 2017-12-08:
Yaml-cpp from Homebrew is not needed on macOS, but hopefully this will make it so it works even if it is, so yes I think so.

dhood, 2017-12-08:
Argh, build machine paths are back (after sourcing `setup.bash` of http://ci.ros2.org/job/ci_packaging_osx/19/ on my macbook):
```
$ echo $DYLD_LIBRARY_PATH
/Users/osrf/jenkins/workspace/ci_packaging_osx/ws/install/opt/rviz_yaml_cpp_vendor/lib:/Users/osrf/jenkins/workspace/ci_packaging_osx/ws/install/opt/rviz_ogre_vendor/lib:/Users/dhood/Downloads/ros2-ardent-oct8-rviz-fix/lib
```

wjwwood, 2017-12-08:
@dhood try with https://github.com/ros2/rviz/pull/147/commits/1c79bbe1c211e586d95dd486d5634da83b050ba9

dhood, 2017-12-08:
hmm. http://ci.ros2.org/job/ci_packaging_osx/19/ in theory includes 5cf4b682da395e098afdb798b9b4df79a877a049 but I don't have `rviz_assimp_vendor_library_path.sh` installed to `share/rviz_assimpl_vendor/environment` like I expected. If I add it there myself then it looks like https://github.com/ros2/rviz/commit/1c79bbe1c211e586d95dd486d5634da83b050ba9 will do what we want. I'm just surprised that it wasn't installed itself.

Perhaps [this condition](https://github.com/ros2/rviz/pull/147/commits/5cf4b682da395e098afdb798b9b4df79a877a049#diff-757a641245eb9af647cbd4b671a66b08R57) is the issue (assimp will be found on the buildfarm machines now, right?)

wjwwood, 2017-12-08:
It's not required if you have assimp installed. `rviz_assimp_vendor` shouldn't do anything on macOS I think.

dhood, 2017-12-08:
I got assimp and yaml-cpp mixed up. I do see the env hook for yaml cpp, and if I use `$AMENT_CURRENT_PREFIX` in it then it works, so I expect https://github.com/ros2/rviz/commit/1c79bbe1c211e586d95dd486d5634da83b050ba9 to have fixed it

wjwwood, 2017-12-08:
That binary is available now, and I just tested it and it works for me.

wjwwood, 2017-12-08:
I'm going to re-close this since I think it's resolved.

closed 2017-12-08 by wjwwood

dhood, 2017-12-08:
yep it is thanks