ros2/rviz #141: fails to transform to the same frame as the fixed frame when there is no tf data

wjwwood, 2017-12-07:
Any of the displays we have so far will complain that they "cannot transform from 'X' frame to 'X' frame", where `X` is something like `map`. If there is tf data which mentions the frame it works, however. Some displays work despite complaining, e.g. Grid, but others do not work in my experience, e.g. PointCloud.

This might be related to the work arounds to avoid the message filters which haven't been fully ported yet.

greimela-si, 2017-12-08:
I don't think this is related to the message filter workaround.
The Grid and PointCloud display just react differently to a tf2::LookupException.
The former just uses (0,0,0) as a fallback position, the latter does not accept new messagesand therefore does not seem to work.

I think this problem should be fixed in tf2, namely that `lookupTransform` should return the identity transformation when source and target frame are equal.

wjwwood, 2017-12-08:
@greimela-si I agree with your assessment. We'll have to look at that next week.

Martin-Idel-SI, 2018-06-14:
Since #298, we have a fix for this bug in RViz (TODO comments in the code to remove the fix when an upstream fix is availabe). Should we close this issue?

closed 2018-06-14 by wjwwood