ros2/rviz #144: Windows support

wjwwood, 2017-12-08:
This is a status report on Windows support for rviz. I'll describe the current state and list some things I've tried and some things that we need to do in order to get Windows fully supported in our "official release" of ROS 2.

First, the current state is this:

- builds through `rviz_common`, but fails to link to `rviz2` (the executable)
- implicitly requires cmake 3.10 (or at least not 3.9)

The linking issue is due to Ogre symbols not being found by the `rviz2_app` executable. The reason (I think) is that `rviz_common` is a static library right now and so is the Ogre library, so it is not getting propagated correctly.

The solution, I believe, is to make both Ogre and `rviz_common`'s library shared. I started doing this last night, but that brought with it some issues with freetype and Ogre as well as needing visibility control (dll import/export) for the API in `rviz_common`. I got that "done" (not fully tested), but then got hung up on some other Ogre issues. Here's the branch for that work in progress:

 https://github.com/ros2/rviz/pull/143

Now, I ended up giving up on getting it in, because even with those issues fixed I would likely need a day or two to get rviz on Windows into mergeable shape for our CI, which would mean:

- actually compiling and linking and running
- no warnings (this includes dependencies like Ogre and the things Ogre builds like freetype)
- proper library paths at runtime (like when using our binary and not building from source)

I simply didn't have time to do that, unfortunately.

I also wanted to point out a few things about the above branch:

- I prevent Ogre from building dependencies
  - I do this because if those dependencies throw warnings then I cannot easily suppress them
- I made Ogre create shared libraries, but the `rviz_ogre_vendor` package's cmake files were modified in a hacky way, so they need to be fixed up
- I made `rviz_common` into a shared library, but I think there might be some lingering issues with the dll import/export stuff I did (again not well tested)
- I didn't try debug builds yet, might be broken again

We could alternatively turn back on Ogre building it's own dependencies, but to do that we'd either need to:

- patch ogre so it suppresses warnings when building its dependencies
- make our build farm ignore these warnings some how
  - might be more work than suppressing them

Oh, and on the topic of suppressing warnings, Windows has a silly thing where I try to suppress all warnings with `/w` but then I get a non-suppressable warning that `/w` overrides the existing command line argument `/W1`. I was frustratingly unable to find where that flag was being introduced and that was another reason I didn't think I could get rid of all of the warnings before the release.

wjwwood, 2017-12-08:
I should have probably posted this on https://github.com/ros2/rviz/issues/133, as it was already talking about getting rviz working on Windows, but since my WIP pr is attached to this one I'll leave it.

Martin-Idel-SI, 2017-12-19:
To help, I tried to build this on both Linux and Windows and here are my findings so far:

For Linux:
- The RenderSystem_GL library cannot be found by CMake since it's missing a 'lib' in front. It's just called "RenderSystem_GL_d.so". Renaming to "libRenderSystem_GL_d.so" seems to do the trick. This is also a problem in the normal Ogre cmake build. 
- After this, the branch builds on Linux until linking in rviz2, where it's missing GLPlugin - I guess this would be inside OgreGLSupport, which is currently commented out in rviz_rendering. Commenting in that line, cmake fails in rviz_rendering. What I find weird is that libOgreGLSupport is still static even when building Ogre dynamically --> maybe an issue in Ogre? Do you know anything about this?

Windows:
- On our system (Win 10 + VS2015) the CMake config of ogre_vendor doesn't work: The FREETYPE_LIBRARIES is a list of objects of the form "optimized;PATH_TO_RELEASE_BUILD;debug;PATH_TO_DEBUG_BUILD", but only the paths must be added here: https://github.com/ros2/rviz/blob/f06898d4d48fb92ccff3c9143a680f5292c1d210/rviz_ogre_vendor/rviz_ogre_vendorConfig.cmake.in#L103
- After that, I get linking issues when linking rviz_rendering to Ogre. Still investigating...

Martin-Idel-SI, 2017-12-20:
I'm adding information as we find it. To keep things organized, I'm just going to edit my findings post.

anhosi, 2017-12-22:
In addition to @Martin-Idel-SI notes above the following is also necessary:
* rviz2 needs to be liked against Ogre: add `rviz_ogre_vendor::OgreMain` and `rviz_ogre_vendor::OgreOverlay` to `target_link_libraries`
* rviz_common tests also need to be linked against Ogre, this includes adding rviz_ogre_vendor as test dependency

Martin-Idel-SI, 2018-01-02:
In addition to @anhosi notes above, I can offer the following next steps:
- The Windows linker issue mentioned above seems to be the same issue as https://github.com/jbeder/yaml-cpp/issues/332 - adding the line `#define YAML_CPP_DLL` into `yaml-config-writer.cpp` and `yaml-config-reader.cpp` (i.e. right before every `#include` of yaml-cpp headers) seems to do the trick.
- This makes `rviz_common` compile cleanly. Next issues are in `rviz_default_plugins`.

Martin-Idel-SI, 2018-01-04:
Running RViz in Windows with VS2015 is not possible yet. The error I get is a `std::regex_error` in the `pluginlib::impl::split` function.  

```
 	rviz_common.dll!std::_Parser<std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > >,char,std::regex_traits<char> >::_Compile() Line 5708	C++
 	rviz_common.dll!std::basic_regex<char,std::regex_traits<char> >::_Reset<std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > > >(std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > > _First, std::_String_const_iterator<std::_String_val<std::_Simple_types<char> > > _Last, std::regex_constants::syntax_option_type _Flags, std::forward_iterator_tag __formal) Line 2560	C++
 	rviz_common.dll!std::basic_regex<char,std::regex_traits<char> >::basic_regex<char,std::regex_traits<char> ><std::char_traits<char>,std::allocator<char> >(const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & _Str, std::regex_constants::syntax_option_type _Flags) Line 2315	C++
>	rviz_common.dll!pluginlib::impl::split(const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & input, const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & regex) Line 45	C++
 	rviz_common.dll!pluginlib::impl::fs::path::path(const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & p) Line 104	C++
 	rviz_common.dll!pluginlib::ClassLoader<rviz_common::Panel>::getPackageFromPluginXMLFilePath(const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & plugin_xml_file_path) Line 566	C++
 	rviz_common.dll!pluginlib::ClassLoader<rviz_common::Panel>::processSingleXMLPluginFile(const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & xml_file, std::map<std::basic_string<char,std::char_traits<char>,std::allocator<char> >,pluginlib::ClassDesc,std::less<std::basic_string<char,std::char_traits<char>,std::allocator<char> > >,std::allocator<std::pair<std::basic_string<char,std::char_traits<char>,std::allocator<char> > const ,pluginlib::ClassDesc> > > & classes_available) Line 705	C++
 	rviz_common.dll!pluginlib::ClassLoader<rviz_common::Panel>::determineAvailableClasses(const std::vector<std::basic_string<char,std::char_traits<char>,std::allocator<char> >,std::allocator<std::basic_string<char,std::char_traits<char>,std::allocator<char> > > > & plugin_xml_paths) Line 320	C++
 	rviz_common.dll!pluginlib::ClassLoader<rviz_common::Panel>::ClassLoader<rviz_common::Panel>(std::basic_string<char,std::char_traits<char>,std::allocator<char> > package, std::basic_string<char,std::char_traits<char>,std::allocator<char> > base_class, std::basic_string<char,std::char_traits<char>,std::allocator<char> > attrib_name, std::vector<std::basic_string<char,std::char_traits<char>,std::allocator<char> >,std::allocator<std::basic_string<char,std::char_traits<char>,std::allocator<char> > > > plugin_xml_paths) Line 99	C++
 	rviz_common.dll!rviz_common::PluginlibFactory<rviz_common::Panel>::PluginlibFactory<rviz_common::Panel>(const QString & package, const QString & base_class_type) Line 68	C++
 	rviz_common.dll!rviz_common::PanelFactory::PanelFactory(const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & node_name) Line 53	C++
 	rviz_common.dll!rviz_common::VisualizationFrame::VisualizationFrame(const std::basic_string<char,std::char_traits<char>,std::allocator<char> > & node_name, QWidget * parent) Line 121	C++
 	rviz_common.dll!rviz_common::VisualizerApp::init(int argc, char * * argv) Line 270	C++
 	rviz2.exe!main(int argc, char * * argv) Line 63	C++
```
The problem is that the regex constructor there is called with `\\`, which doesn't work.
Even in the most minimal example shows that this regex doesn't work:

```
#include "stdafx.h"
#include <regex>

int main()
{
    std::regex re("\\");
    return 0;
}
```

fails. Is this a problem in the pluginlib?

Martin-Idel-SI, 2018-01-08:
On to Visual Studio 2017:

- At the moment I get a lot of errors of the type 
```
5>c:\dev\ros2\install\include\pluginlib\./impl/filesystem_helper.hpp(46): error C2039: 'filesystem': is not a member of 'std' (compiling source file C:\jw\7ac31f0\src\rviz_common\src\rviz_common\display_factory.cpp) [C:\jw\7ac31f0\build\rviz_common\rviz_common.vcxproj]
```
I haven't looked at the code, as this is once again the pluginlib, but it seems that even with VS2017, filesystem is indeed not a member of "std": https://developercommunity.visualstudio.com/content/problem/116454/c-filesystem-still-experimental-even-with-stdclate.html

wjwwood, 2018-01-08:
The issue with filesystem on Windows is also being worked on here: https://github.com/ros/pluginlib/pull/96

Martin-Idel-SI, 2018-01-09:
The issue with the filesystem actually fixes also the regex error (I beliebe, because it doesn't use the workaround?) on our system. This means we get a compiling build and green tests on Linux, VS2015 and VS2017.

On to running in Windows:
- When starting, RViz doesn't find any plugins. This is a problem with the plugins-description.xml on Windows. It's geared towards Linux reading
```
<library path="librviz_default_plugins">
```
The "lib" prefix is Unix specific and the libraries are apparently searched in `install\lib`, while the dlls on Windows are located in `install\bin`. The following line works on Windows:
```
<library path="../bin/rviz_default_plugins">
```
- When the plugins are found, a grid is again visible in the main Window. However, the display panel doesn't work. Investigating...

heiko134470, 2018-01-17:
Hi wjwwood,

I have a question about rviz of windows, until when can we use rviz of ros2 on windows?

wjwwood, 2018-01-17:
> I have a question about rviz of windows, until when can we use rviz of ros2 on windows?

Until it's ready. Right now, I'm waiting on this main pull request to get in shape so that it can be merged. On my side I was just going to try and suppress the compiler warnings from the external dependencies we build. But there are also compiler warnings in the rviz code that need to be fixed, and I haven't planned any time to spend on that.

Until then, you can build from source using the branch we're developing on.

wjwwood, 2018-01-23:
There is still a few issues with windows but we’re following up on them (like the pluginlib pr https://github.com/ros/pluginlib/pull/97). But we’re tracking those separately. I’m going to close this with #175’s merging.

closed 2018-01-23 by wjwwood