ros2/rviz #146: RViz throws a Poco::SystemException on shutdown

greimela-si, 2017-12-08:
When closed, RViz terminates with aegmentation fault and the message
```
terminate called after throwing an instance of 'Poco::SystemException'
  what():  System exception
```

This also results in multiple test failures in `rviz_common`:
- rviz_common_property_test
- point_cloud2_display_test 
- point_cloud_transformers_test

greimela-si, 2017-12-11:
This issues seems to occur if one has an old version of Poco (e.g. 1.3.x) installed on the system.
In that case the `poco_vendor` package distributed with ros2 will build a newer version (1.4.1) from source.
However, the Ogre build in `rviz_ogre_vendor` prefers to link with the system Poco, resulting in the observed crashes when shutting down RViz.

To fix this we have to prioritize linking with the poco provided by the ros2 `poco_vendor` package.

wjwwood, 2017-12-12:
That sounds like the right problem, question is how to make Ogre use the one we want 😄.

@dirk-thomas maybe this is the segfault on close you were seeing?

dirk-thomas, 2017-12-13:
> @dirk-thomas maybe this is the segfault on close you were seeing?

No, this is different. In my case I get the following backtrace on close:

```
#0  malloc_consolidate (av=av@entry=0x7ffff0d8bb20 <main_arena>) at malloc.c:4175
#1  0x00007ffff0a48cde in _int_malloc (av=av@entry=0x7ffff0d8bb20 <main_arena>, bytes=bytes@entry=2808) at malloc.c:3450
#2  0x00007ffff0a4bdca in __libc_calloc (n=<optimized out>, elem_size=<optimized out>) at malloc.c:3236
#3  0x00007fffef7ff9e3 in ?? () from /usr/lib/nvidia-384/libGL.so.1
#4  0x00007fffef7f875a in ?? () from /usr/lib/nvidia-384/libGL.so.1
#5  0x00007fffef7f8fdc in ?? () from /usr/lib/nvidia-384/libGL.so.1
#6  0x00007fffeac6809e in ?? () from /usr/lib/nvidia-384/libnvidia-glcore.so.384.90
#7  0x00007fffeac682fb in ?? () from /usr/lib/nvidia-384/libnvidia-glcore.so.384.90
#8  0x00007fffeac68389 in ?? () from /usr/lib/nvidia-384/libnvidia-glcore.so.384.90
#9  0x00007fffead5dce3 in ?? () from /usr/lib/nvidia-384/libnvidia-glcore.so.384.90
#10 0x00007fffead62d2a in ?? () from /usr/lib/nvidia-384/libnvidia-glcore.so.384.90
#11 0x00007fffead1b3ea in ?? () from /usr/lib/nvidia-384/libnvidia-glcore.so.384.90
#12 0x00007fffeac8ca61 in ?? () from /usr/lib/nvidia-384/libnvidia-glcore.so.384.90
#13 0x00007fffeac656cc in ?? () from /usr/lib/nvidia-384/libnvidia-glcore.so.384.90
#14 0x00007fffef7a03f1 in ?? () from /usr/lib/nvidia-384/libGL.so.1
#15 0x00007fffef7d39a4 in ?? () from /usr/lib/nvidia-384/libGL.so.1
#16 0x00007fffef7d8739 in ?? () from /usr/lib/nvidia-384/libGL.so.1
#17 0x00007fffef849234 in ?? () from /usr/lib/nvidia-384/libGL.so.1
#18 0x00007ffff7de7e05 in _dl_fini () at dl-fini.c:240
#19 0x00007ffff0a00ff8 in __run_exit_handlers (status=0, listp=0x7ffff0d8b5f8 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true) at exit.c:82
#20 0x00007ffff0a01045 in __GI_exit (status=<optimized out>) at exit.c:104
#21 0x00007ffff09e7837 in __libc_start_main (main=0x493496 <main(int, char**)>, argc=1, argv=0x7fffffff5b78, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffff5b68)
    at ../csu/libc-start.c:325
#22 0x0000000000492ca9 in _start ()
```

clalancette, 2021-05-26:
We no longer use Poco at all, so this particular issue shouldn't occur anymore.  I'm going to close this out, but do feel free to open a new issue if you are still running into issues.

closed 2021-05-26 by clalancette