ros2/rviz #150: Can't seem to visualize any topics on Linux/AMD64

clalancette, 2017-12-08:
I am trying a simple test where I run the dummy robot using:

```
launch /opt/ros/ardent/share/dummy_robot_bringup/launch/dummy_robot_bringup.p
```

And then starting up rviz2.  Once I start up rviz2, I would expect that I could click the `Add` button, and add a topic by name to start visualizing it.  However, all topics are showing up as unvisualizable.  I would think that the `/joint_states`, `/tf`, and others would be visualizable.

Karsten1987, 2017-12-08:
This worked for me, at least on my mac. See the tutorial at the bottom: https://github.com/ros2/ros2/wiki/dummy-robot-demo

When changing the global reference frame, do you see the available links of TF?

wjwwood, 2017-12-08:
> and add a topic by name to start visualizing it

@Karsten1987 I think he just means the "add display by topic" doesn't work. I believe that if he adds the displays by type and configures them it should just work.

----

@clalancette I might be wrong, but I think the TF display never showed up on the add display by topic dialogue. There's no display for joint states right now. So maybe there really aren't any visualizable topics yet.

If you have some point cloud data you could try to add by topic then. I'll try that locally and report back.

wjwwood, 2017-12-08:
Nah, it looks broken, as I couldn't add by topic a `sensor_msgs/PointCloud.msg` topic.

clalancette, 2017-12-08:
> @Karsten1987 I think he just means the "add display by topic" doesn't work. I believe that if he adds the displays by type and configures them it should just work.

Right.

> @clalancette I might be wrong, but I think the TF display never showed up on the add display by topic dialogue. There's no display for joint states right now. So maybe there really aren't any visualizable topics yet.

Ah, maybe that is it.  I just sort of expected that, but maybe it doesn't work.

Separately but related, if I start up the dummy_robot_demo, then startup rviz2, I can never visualize the robot (even when following the instructions).  However, if I startup rviz2 first, then startup the demo, things work.

nuclearsandwich, 2017-12-08:
@clalancette was this testing Linux debs on AMD64?

clalancette, 2017-12-08:
Yeah, it was using the debs.

dhood, 2017-12-08:
I saw communication issues too @clalancette but not when I switched to opensplice (on os x using archives), can you try that?

wjwwood, 2017-12-08:
> Separately but related, if I start up the dummy_robot_demo, then startup rviz2, I can never visualize the robot (even when following the instructions). However, if I startup rviz2 first, then startup the demo, things work.

Can you open another issue and update the title of this one?

As I see it there are two unrelated issues:

- add display by topic is broken
- displays don't always show data unless rviz is started first (maybe only with fastrtps)

clalancette, 2017-12-08:
> I saw communication issues too @clalancette but not when I switched to opensplice (on os x using archives), can you try that?

@dhood Yep, switching to opensplice fixed that particular problem.

> Can you open another issue and update the title of this one?

Yeah, I'll do that.

greimela-si, 2017-12-11:
We have to list the message types a display is able to visualize in the `plugins_description.xml`. 
This is how RViz determines the visualizable topics.

We are able to fix this as soon as the displays are loaded using the pluginlib.
I integrated the relevant change for the point cloud displays in #153:
https://github.com/bsinno/rviz/blob/52129fefa60a4bef1e96c5aed744b2f213b5f076/rviz_default_plugins/plugins_description.xml#L23

wjwwood, 2017-12-12:
@greimela-si makes sense, it's possible we could add that information to the `addBuiltInClass()` call we make to register them without going through the plugin:

https://github.com/ros2/rviz/blob/fa0a5ed87a5ba367914eca34d37799e8deba2028/rviz_common/src/rviz_common/display_factory.cpp#L79-L82

If not, it will be resolved when we can move them all.

greimela-si, 2017-12-18:
FYI: Add display by topic does now work for PointClouds since #153 has been merged.

wjwwood, 2017-12-18:
Great, I'll close this since we don't intend to leave any of the displays unported and there fore this will not be an issue moving forward.

closed 2017-12-18 by wjwwood