ros2/rviz #151: Displays don't always show data unless rviz is started first (with Fast-RTPS)

clalancette, 2017-12-08:
I'm trying to visualize the demo here: https://github.com/ros2/ros2/wiki/dummy-robot-demo .  I'm on Linux/AMD64, using the Ardent debs and using Fast-RTPS.  If I start up the `dummy_robot_demo` first, then startup rviz2, I never seem to get the visualization properly.  If I start up rviz2 first, then `dummy_robot_demo`, things seem to work and I can see the visualization.  Switching to Opensplice also seems to make it work.

dhood, 2017-12-08:
For me (on OS X from archives) this was just generally flaky with fastrtps, the order didn't matter. When I had data, it would cut out after a while. I suspect general throughput issues.

gaoethan, 2017-12-22:
@clalancette it seems that the rviz2 finally works for you, can I check how you prepare the rviz2 to run normally ?

1. Binary or build from rviz2 source code ?
2. The way to start rviz2 with "ros2 run" isn't available, so what's your procedure to start rviz2 ? 

thanks.

clalancette, 2017-12-22:
In this case, I was running from a binary.  I just ran the executable directly, i.e. `/opt/ros/ardent/bin/rviz2`

clalancette, 2021-05-26:
This isn't a problem for me anymore, so I'm going to close this one out.

closed 2021-05-26 by clalancette