ros2/rviz #152: missing break in switch-case in config.cpp?

fmauch, 2017-12-10:
Since in the CMakeLists.txt the compiler is set to pedandic and Werror, build fails on rviz_common/config.cpp:145 as there is no break in the case section when using gcc7 (gc 7.2.1 used here):

```/home/felix/Downloads/ros2/src/ros2/rviz/rviz_common/src/rviz_common/config.cpp: In member function ‘void rviz_common::Config::copy(const rviz_common::Config&)’:
/home/felix/Downloads/ros2/src/ros2/rviz/rviz_common/src/rviz_common/config.cpp:145:9: error: this statement may fall through [-Werror=implicit-fallthrough=]
         }
         ^
/home/felix/Downloads/ros2/src/ros2/rviz/rviz_common/src/rviz_common/config.cpp:147:5: note: here
     case Value:
     ^~~~
At global scope:
cc1plus: error: unrecognized command line option ‘-Wno-gnu-zero-variadic-macro-arguments’ [-Werror]
cc1plus: all warnings being treated as errors```

I know that gcc7 is not officially supported, but this might still be unintentional? Otherwise a statement suppressing this warning or at least a comment why there is no break there would be nice I guess.

mjbogusz, 2017-12-10:
I've encountered that too and I think this actually is a bug inherited from the mainline. GCC7 simply added `implicit-fallthrough` warning to it's `-Wextra`.

My understanding is that there __should__ be a `break;` statement, as falling through to the last case will erroneously set the type of the Config object to `Value` (within `.setValue()`).

Another issue is the `unrecognized command line option ‘-Wno-gnu-zero-variadic-macro-arguments’`: this is solved by explicitly setting the project type in `CMakeLists.txt`:
```cmake
project(rviz_common CXX)
```
(solution found [here](https://stackoverflow.com/questions/25451254/getting-cmake-check-cxx-compiler-flag-to-work#25451834))

---

The two issues above are in fact unrelated to each other - I'm not sure whether it would be better to open a PR for both of them at once or separately?

mjbogusz, 2017-12-11:
After fixing those issues I've hit another wall:
```
In file included from /usr/include/c++/7.2.1/ext/string_conversions.h:41:0,
                 from /usr/include/c++/7.2.1/bits/basic_string.h:6349,
                 from /usr/include/c++/7.2.1/string:52,
                 from /home/mjbogusz/documents/ros2/src/ros2/rviz/rviz2/src/main.cpp:31:
/usr/include/c++/7.2.1/cstdlib:75:15: fatal error: stdlib.h: No such file or directory.
 #include_next <stdlib.h>
               ^~~~~~~~~~
compilation terminated.
```

After setting `CMAKE_VERBOSE_MAKEFILE` to true and analyzing the compilation command I've noticed this little gem:
```
-isystem /usr/lib64/cmake/yaml-cpp/../../../include
```
That effectively makes `/usr/include/stdlib.h` go up on the list of possible `stdlib.h` files to include from `cstdlib` and therefore `#include_next` has no valid target ([this](https://stackoverflow.com/a/46598311/3217805) helped me understand that).

I've traced it down to `rviz_yaml_cpp_vendor` triggering environment hooks and setting additional include directories even if `yaml-cpp` is found on the system. I've fixed that by wrapping the whole download/build/hook/package block in an if statement:
```cmake
find_package(yaml-cpp QUIET)
if(NOT yaml-cpp_FOUND)
    (...)
else()
    ament_package()
endif()
```
To ensure proper linking of `yaml-cpp` however the libraries for `rviz_yaml_cpp_vendor` had to be set so I've added another cmake 'extras' file doing just that.

I'll create a PR soon.

racko, 2017-12-14:
Looking forward to your PR :)

Theosakamg, 2017-12-14:
The break is needed ?

From https://developers.redhat.com/blog/2017/03/10/wimplicit-fallthrough-in-gcc-7/
I use regex /* FALLTHRU */

racko, 2017-12-14:
I created the following pull requests to fix the issues listed here: #157, #158, #159.

mjbogusz, 2017-12-15:
@Theosakamg yes, the break is indeed needed there. It's an upstream bug.

@racko I have a slightly more complete soliution for the `yaml-cpp` thingy but I want to clean it up and test it first.

wjwwood, 2017-12-16:
Since the originally reported issue has been resolved in #158 I'm going to close this.

closed 2017-12-16 by wjwwood