ros2/rviz #154: Some tests are currently disabled

greimela-si, 2017-12-12:
Tests that did not run on the ROS2 CI (mostly rendering related) are currently commented out in the code.
The reason is supposedly the missing display on the Jenkins slave.

Ideally the ROS2 Jenkins setup should be able to run the tests. 
Our local Jenkins currently executes the tests using the Jenkins Xvfb plugin.

@wjwwood: Is it possible to update the `ci.ros2.org` setup to run these tests?

wjwwood, 2017-12-13:
Yeah, sorry this was going to be mentioned in my debrief issue about that release, but I haven't gotten around to make that issue yet.

> @wjwwood: Is it possible to update the ci.ros2.org setup to run these tests?

No idea 😄.  Might be hard since they are running in docker and the linux host for docker is also headless. Maybe @j-rivero or @scpeters or @sloretz have some ideas about this given their experience testing gazebo.

scpeters, 2017-12-13:
We run GUI and GPU tests with nvidia-docker on some machines on build.osrfoundation.org. It's not too hard to get it configured, but it does require an nvidia GPU. I have tried with little success to get intel GPU's working with CI, but their power-save features are too strong; I can't get the GPU to work without a monitor physically plugged in.

I've had much fewer issues with the lack of an external display when running GUI and GPU tests on our mac machines, though I do have to make sure the XQuartz app is running for the current user (each user has a distinct X server). I made the XQuartz app one of the startup items for our jenkins user, but I always make sure to manually log in to that user whenever I reboot the machine after software updates.

sloretz, 2017-12-13:
Is this related to needing an X server to get a render context for opengl? Running gazeob in docker with rendering requires an X server running on the host, with a socket to it mounted as a volume and some related environment variables ([see this script for an example](https://bitbucket.org/sloretz/dev_docker/src/36ec7f3b18929511568d17947659e08709e67017/run_ws.bash?at=develop&fileviewer=file-view-default)).

I haven't investigated, but supposedly EGL solves the problem.
https://devblogs.nvidia.com/parallelforall/egl-eye-opengl-visualization-without-x-server/
https://www.khronos.org/egl

wjwwood, 2017-12-13:
In my experience rviz doesn't need X11 to work on macOS. Though that might be our newer version of ogre (`1.10.10` + some commits, i.e. we're using some commit from master until `1.10.11` is released).

scpeters, 2017-12-13:
do you have a formula for ogre 1.10? I was testing one but it had some runtime problems with gazebo that I haven't resolved

wjwwood, 2017-12-13:
Nope, we build it as one of our packages right now:

https://github.com/ros2/rviz/tree/ros2/rviz_ogre_vendor

This is because we need it on Windows too, and until we can use a released version without patches, it's easier to iterate on it as one of our packages we build.

greimela-si, 2017-12-13:
> In my experience rviz doesn't need X11 to work on macOS
It does not specifically require X11, but some kind of display server is necessary to get an OpenGL context.
A virtual framebuffer (e.g. xvfb) should be enough on a Linux slave, but I don't know if something similar exists in the macOS ecosystem.
On Windows a physical display seems to be required.

Martin-Idel-SI, 2018-05-07:
I'm going to add some information I found when investigating some issues on Windows:

- Here is a question listing prerequisites to running display tests on Windows https://stackoverflow.com/questions/19441324/how-to-run-gui-tests-on-a-jenkins-windows-slave-without-remote-desktop-connectio
- This implies that running as System user only will probably not be enough. There needs to be some kind of "User" to have a GUI
- For us, we also had problems running Jenkins via a Windows system service, we succeeded by using a .bat file and autostart it. This might also be a problem.