ros2/rviz #155: Build fail on 'rviz_common'

Theosakamg, 2017-12-13:
## Bug report

**Required Info:**

- Operating System:
  - Debian Testing/SID
- Installation type:
  - from source
- Version or commit hash:
  - f00807441bac7efdfef337fe0cd8ecc4a7ef1d56
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - NA

#### Steps to reproduce issue

```
Process package 'rviz_assimp_vendor' with context:
--------------------------------------------------------------------------------
 source_space => /home/micky/ros_workspace/toto/src/ros2/rviz/rviz_assimp_vendor
  build_space => /home/micky/ros_workspace/toto/build_isolated/rviz_assimp_vendor
install_space => /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor
   make_flags => -j4, -l4
  build_tests => False
--------------------------------------------------------------------------------
+++ Building 'rviz_assimp_vendor'
Running cmake because arguments have changed.
==> '. /home/micky/ros_workspace/toto/build_isolated/rviz_assimp_vendor/cmake__build.sh && /usr/bin/cmake /home/micky/ros_workspace/toto/src/ros2/rviz/rviz_assimp_vendor -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor' in '/home/micky/ros_workspace/toto/build_isolated/rviz_assimp_vendor'
-- The C compiler identification is GNU 7.2.0
-- The CXX compiler identification is GNU 7.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.4.0 (/home/micky/ros_workspace/ament_ws/install_isolated/ament_cmake/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.3", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Configuring done
-- Generating done
-- Build files have been written to: /home/micky/ros_workspace/toto/build_isolated/rviz_assimp_vendor
==> '. /home/micky/ros_workspace/toto/build_isolated/rviz_assimp_vendor/cmake__build.sh && /usr/bin/make -j4 -l4' in '/home/micky/ros_workspace/toto/build_isolated/rviz_assimp_vendor'
+++ Installing 'rviz_assimp_vendor'
==> '. /home/micky/ros_workspace/toto/build_isolated/rviz_assimp_vendor/cmake__install.sh && /usr/bin/make install' in '/home/micky/ros_workspace/toto/build_isolated/rviz_assimp_vendor'
Install the project...
-- Install configuration: ""
-- Execute custom install script
-- Symlinking: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/share/ament_index/resource_index/package_run_dependencies/rviz_assimp_vendor
-- Symlinking: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/share/ament_index/resource_index/parent_prefix_path/rviz_assimp_vendor
-- Symlinking: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/share/rviz_assimp_vendor/environment/ament_prefix_path.sh
-- Symlinking: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/share/rviz_assimp_vendor/environment/path.sh
-- Symlinking: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/share/rviz_assimp_vendor/local_setup.bash
-- Symlinking: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/share/rviz_assimp_vendor/local_setup.sh
-- Symlinking: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/share/rviz_assimp_vendor/local_setup.zsh
-- Symlinking: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/share/ament_index/resource_index/packages/rviz_assimp_vendor
-- Symlinking: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/share/rviz_assimp_vendor/cmake/rviz_assimp_vendor-extras.cmake
-- Symlinking: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/share/rviz_assimp_vendor/cmake/rviz_assimp_vendorConfig.cmake
-- Symlinking: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/share/rviz_assimp_vendor/cmake/rviz_assimp_vendorConfig-version.cmake
-- Symlinking: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/share/rviz_assimp_vendor/package.xml
-- [ament] Deploying: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/local_setup.bash
-- [ament] Deploying: /home/micky/ros_workspace/toto/install_isolated/rviz_assimp_vendor/local_setup.sh
[... 81 lines cut ...]
-- Found tinyxml_vendor:  (/home/micky/ros_workspace/toto/install_isolated/tinyxml_vendor/share/tinyxml_vendor/cmake)
-- Found tf2: 0.8.0 (/home/micky/ros_workspace/toto/install_isolated/tf2/share/tf2/cmake)
-- Found tf2_geometry_msgs: 0.8.0 (/home/micky/ros_workspace/toto/install_isolated/tf2_geometry_msgs/share/tf2_geometry_msgs/cmake)
-- Found urdf: 2.0.0 (/home/micky/ros_workspace/toto/install_isolated/urdf/share/urdf/cmake)
-- Found TinyXML: /usr/lib/x86_64-linux-gnu/libtinyxml.so  
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
library_abs
    linked by target "rviz_common" in directory /home/micky/ros_workspace/toto/src/ros2/rviz/rviz_common

-- Configuring incomplete, errors occurred!
See also "/home/micky/ros_workspace/toto/build_isolated/rviz_common/CMakeFiles/CMakeOutput.log".

<== Command '. /home/micky/ros_workspace/toto/build_isolated/rviz_common/cmake__build.sh && /usr/bin/cmake /home/micky/ros_workspace/toto/src/ros2/rviz/rviz_common -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/micky/ros_workspace/toto/install_isolated/rviz_common' failed in '/home/micky/ros_workspace/toto/build_isolated/rviz_common' with exit code '1'
<== Command '. /home/micky/ros_workspace/toto/build_isolated/rviz_common/cmake__build.sh && /usr/bin/cmake /home/micky/ros_workspace/toto/src/ros2/rviz/rviz_common -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_INSTALL_PREFIX=/home/micky/ros_workspace/toto/install_isolated/rviz_common' failed in '/home/micky/ros_workspace/toto/build_isolated/rviz_common' with exit code '1'

```

#### Expected behavior

#### Actual behavior

#### Additional information

Thank for you help.

Theosakamg, 2017-12-13:
Work if I remove assimp installed on the system.

closed 2017-12-13 by Theosakamg

wjwwood, 2017-12-13:
Hmm, ideally it would work even with the system assimp installed, but it's good you found a work around in the meantime.

This "vendor" package for assimp is hopefully temporary and we'll be using the system provided version asap.