ros2/rviz #168: Wiki links refered in the package.xml of rviz2 are unreachable

gaoethan, 2017-12-22:
Wiki links in the **package.xml** of rviz2 components are not valid, it indicates the error like "This page does not exist yet" when trying to access them in the browser, for examples:

[rviz2](https://github.com/ros2/rviz/blob/ros2/rviz2/package.xml#L15)
[rviz_default_plugins](https://github.com/ros2/rviz/blob/ros2/rviz_default_plugins/package.xml#L14)
[rviz_comm](https://github.com/ros2/rviz/blob/ros2/rviz_default_plugins/package.xml#L14)
[rviz_rendering](https://github.com/ros2/rviz/blob/ros2/rviz_rendering/package.xml#L14)

gaoethan, 2017-12-22:
Maybe it's on the way to prepare them, but it makes users confused in a sense, so... mark the issue here, thanks !

Martin-Idel-SI, 2018-06-28:
This should no longer be a problem since updating the link to point to the README.

closed 2018-06-28 by wjwwood