ros2/rviz #171: Array delete misuse

yechun1, 2018-01-08:
~Any uses delete[] on <temporary>.mContent which is not an array.

selection_manager.cpp::1125
```
1125 rend->getUserObjectBindings().setUserAny("pick_handle", Ogre::Any(handle_));
1138 object->getUserObjectBindings().setUserAny("pick_handle", Ogre::Any(handle));
```

OgreAny.h
```
 // Freeing function reset deletes parameter this->mContent
79        virtual ~Any()
80        {
81            reset(); 

```

```
// Freeing function deallocateBytes deletes parameter (void *)this->mContent
131        void reset()
132        {
133            OGRE_DELETE_T(mContent, placeholder, MEMCATEGORY_GENERAL);
```

OgreMemoryStdAlloc.h
```
75                static inline void deallocateBytes(void* ptr)
80                        delete[]((unsigned char*)ptr);
```

wjwwood, 2018-01-08:
@yechun1 Please reference code rather than copy-pasting it into the issue. This appears to be an issue with Ogre's headers, not how we're using it, is that correct? In which case, I'd ask you to open an issue on the Ogre issue tracker:

https://github.com/ogrecave/ogre/issues

If it is something you think we can address in our usage of Ogre, then please suggest that, but until then I'm going to close this issue. I can reopen it in the future if needed.

closed 2018-01-08 by wjwwood

yechun1, 2018-08-16:
@wjwwood 
The issue was caused by ORGE and had been fixed in commit [dc990c5](https://github.com/OGRECave/ogre/commit/dc990c5) 

Suggest to upgrade Ogre dependent package with newer version.