ros2/rviz #176: Visual Studio 14 or 15?

heiko134470, 2018-01-17:
Hi my friends,

in the page of ROS2-installation(https://github.com/ros2/ros2/wiki/Windows-Development-Setup) the author has already emphasized the following:
**First you'll need Visual Studio 14.0 2015 Update 3 installed. With Visual C++ selected during the installation 
2017 is newer but is known to have some issues that have not been resolved.**

However, you suggest that we should install Visual Studio 15 2017, if we want to use rviz.

Those both are confusing me.

Could anybody explain that?

Thanks in advance!

greimela-si, 2018-01-17:
Hi Heiko,

yes, it looks like the documentation is not clear in this aspect.

Right now, Visual Studio 14 2015 is the preferred version when building ROS2 and any project in this context. The CI builds on ci.ros2.org also use this version to build and test on Windows.

Visual Studio 15 2017 does also work, but we experienced some problems with it in the past.

Please also notice that the Windows build is currently broken, that issue is tracked in #144.

heiko134470, 2018-01-17:
This means, I can't use rviz now? Until when can you fix all bugs?

greimela-si, 2018-01-17:
Unfortunately not on Windows, no...
@wjwwood might have some details regarding the estimated time.

closed 2018-01-17 by wjwwood