ros2/rviz #179: RViz2 binary linked against libyaml-cpp in build folder

greimela-si, 2018-01-24:
I noticed that the `rviz2` binary on my system is linked against the `libyaml-cpp.so.0.5` in the build folder, and not the one in the install folder:

```
$ ldd rviz2 | grep yaml 
libyaml-cpp.so.0.5 => /[..]/rviz2_ws/build/rviz_yaml_cpp_vendor/yaml_cpp-86ae3a5-prefix/src/yaml_cpp-86ae3a5-build/libyaml-cpp.so.0.5
```

The reason seems to be the `RPATH` that is included in the binary. This variable is used preferrably over `LD_LIBRARY_PATH` to find libraries and includes the yaml-cpp `build/` folder instead of the `install/` folder.

It is enabled using this line:

https://github.com/ros2/rviz/blob/e39c5593f1a989310d499b44dae9e50164509a75/rviz2/CMakeLists.txt#L97

Without this line, libyaml-cpp is correctly found in the install directory, but now rviz_common can not be found anymore.
This is because `rviz_ws/install/lib` is not added to the `LD_LIBRARY_PATH`.

I suggest to remove the `RPATH` from the rviz2 binary and fix the `LD_LIBRARY_PATH`. That way all libraries are loaded using the same source.

Proposed TODOs:
- [ ] Remove the RPATH from the rviz2 binary
- [ ] If required: Enable the RPATH on macOS
- [ ] Add `rviz_ws/install/lib` to the LD_LIBRARY_PATH on setup.sh

wjwwood, 2018-01-30:
> This is because `rviz_ws/install/lib` is not added to the `LD_LIBRARY_PATH`.

I would need to see more details about this, because it should be added to the library path, as here:

https://github.com/ament/ament_package/blob/master/ament_package/template/environment_hook/library_path.sh

If that were not working, lots of other things would be broken.

greimela-si, 2018-02-01:
Sure, here are some more details.

My current LD_LIBRARY_PATH after sourcing both the ros2 and rviz workspace:
```
$ env | grep LD_LIBRARY_PATH
LD_LIBRARY_PATH=
/usr/lib:
/home/greimela/projects/ros2/rviz2_ws/install/opt/rviz_yaml_cpp_vendor/lib:
/home/greimela/projects/ros2/rviz2_ws/install/opt/rviz_ogre_vendor/lib:
/home/greimela/projects/ros2/rviz2_ws/install/opt/rviz_assimp_vendor/lib:
/home/greimela/projects/ros2/ros2_ws/install/lib
```
Notice that there is no `rviz2_ws/install/lib` in there.

The `RPATH` of my rviz2 binary:
```
$ readelf -d ./install/bin/rviz2  | grep RPATH
 0x000000000000000f (RPATH)              Library rpath: [
/home/greimela/projects/ros2/ros2_ws/install/lib:
/home/greimela/projects/ros2/rviz2_ws/install/lib:
/home/greimela/projects/ros2/rviz2_ws/install/opt/rviz_ogre_vendor/lib:
/home/greimela/projects/ros2/rviz2_ws/build/rviz_yaml_cpp_vendor/yaml_cpp-86ae3a5-prefix/src/yaml_cpp-86ae3a5-build:
/home/greimela/projects/ros2/rviz2_ws/install/opt/rviz_assimp_vendor/lib]
```
Here the `rviz2_ws/install/lib` can be found.

If I understand the [ld man page](https://linux.die.net/man/8/ld.so) correctly the loader first searches for shared libraries using the hard wired `RPATH` of the binary. If it is not able to find the library there it uses the `LD_LIBRARY_PATH`.

However, this problem only occurs when compiling rviz in a separate workspace, since `ros2_ws/install/lib` is always added correctly.
Therefore we could postpone this issue for now.

greimela-si, 2018-06-14:
This seems to be solved, even though I don't know what fixed it.

closed 2018-06-14 by greimela-si