ros2/rviz #180: RViz segfault on macOS (v. 10.13.2) with latest master

botteroa-si, 2018-01-24:
Following the change from static to dynamic linking, on Apple (on Linux and Windows this problem doesn't occur) RViz crashes at startup with a segfault. 
The problem was traced back to the `params["externalWindowHandle"] = Ogre::StringConverter::toString(window_id);` line, in render_system.cpp (line 428). When it is commented out, the segfault doesn't happen, but, of course, RViz starts with the render window detached from the application window. 

Proposed Solution: apparently, updating to the latest version of Ogre fixes the problem.

Martin-Idel-SI, 2018-01-29:
Upon closer inspection, the error still persists with Ogre 1.10.11. However, it sometimes works, so maybe there is a race condition somewhere. We'll investigate.

botteroa-si, 2018-02-06:
After simply deleting the install and build folders and building everything again, RViz works fine again, and no segfault occurs. It is not clear what the reason behind this behaviour is.

wjwwood, 2018-02-07:
@botteroa-si, I believe @Martin-Idel-SI said it was possibly a race condition, so maybe you just got lucky after rebuilding? Can you confirm that it never happens anymore?

botteroa-si, 2018-02-07:
@wjwwood, before rebuilding the segfault would happen 95% of the times I tried to launch RViz. After rebuilding yesterday afternoon, it hasn't happened again yet, and I have launched and used RViz quite some times since then.

botteroa-si, 2018-02-07:
@wjwwood, I have now deleted the install and build folders and rebuilt three times, and all three times I have experienced no segfault.

wjwwood, 2018-02-07:
Cool, thanks for following up. In that case, I'll close this until someone else runs into it.

closed 2018-02-07 by wjwwood