ros2/rviz #185: pointcloud display in rviz2 for ros 2

sagniknitr, 2018-01-30:
I have created a demo node which publishes the point cloud data from .pcd file in a senosr_msgs/PointCloud2 format.The rviz tool for ros 2 is able to detect the pointcloud topic name, but it is coming under unvisualizable topic.

Is there any way where I can visualize the sensor_msgs/PointCloud data in rviz display for ROS2 environment ?

greimela-si, 2018-01-30:
This issue should be fixed on the current master.
Are you using the release version or are you building from source?

As a workaround you can add a PointCloud display by name and then fill in the topic name on the left side.

[image]

sagniknitr, 2018-01-30:
@greimela-si I am using the ardent branch of the rviz for ros 2.Is the issue fixed in the rviz 2.0.0 release ?

sagniknitr, 2018-01-30:
[image]

I have attached my rviz window.As you can see, the pointcloud2 display window is receiving the messages. but no display is coming in the viewing area.

jeising, 2018-01-30:
You would need to publish a transform e.g.:
`ros2 run tf2_ros static_transform_publisher -- 0 0 0 0 0 0 map pointcloud_frame`

sagniknitr, 2018-01-30:
@jeising Is the `pointcloud_frame` you mentioned in the command above is same as the` header.frame_id `of the PCL message ?

sagniknitr, 2018-01-30:
[image]

@jeising Did what you mentioned.Still no output.

Martin-Idel-SI, 2018-01-31:
Two more questions:
- Can you switch the Style from "Flat Squares" to "Points"? Maybe there is a rendering problem and points is the easiest. 
- Did you ever try to publish PointCloud instead of PointCloud2 and if yes, did you see anything?

greimela-si, 2018-01-31:
We were able reproduce the bug, here is the bugfix PR: #189

sagniknitr, 2018-02-01:
@greimela-si Thanks for the PR:#189. It solves my problem.

closed 2018-02-01 by wjwwood