ros2/rviz #192: rviz not building on RPi3 due to Ogre being static library (?)

Ekci, 2018-02-01:
## Bug report

- Operating System:
  - RPi3 Ubuntu Mate 16.04
- Installation type:
  - binaries
- Version or commit hash:
  - following https://github.com/ros2/ros2/wiki/Linux-Development-Setup (version from 01.02.2018)
- DDS implementation:
  - following https://github.com/ros2/ros2/wiki/Linux-Development-Setup (version from 01.02.2018)

#### Steps to reproduce issue
I am following building manual. The only change to it is that I use:
```src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install --skip-packages ros1_bridge --make-flags -j1```

#### Expected behavior
Build whole ROS2
#### Actual behavior
Process breakes with rviz_rendering error:
```
Process package 'rviz_rendering' with context:
--------------------------------------------------------------------------------
 source_space => /home/rpi/ros2_ws/src/ros2/rviz/rviz_rendering
  build_space => /home/rpi/ros2_ws/build/rviz_rendering
install_space => /home/rpi/ros2_ws/install
   make_flags => -j1
  build_tests => True
--------------------------------------------------------------------------------
+++ Building 'rviz_rendering'
==> '. /home/rpi/ros2_ws/build/rviz_rendering/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/home/rpi/ros2_ws/build/rviz_rendering'
==> '. /home/rpi/ros2_ws/build/rviz_rendering/cmake__build.sh && /usr/bin/make -j1' in '/home/rpi/ros2_ws/build/rviz_rendering'
[  3%] Automatic moc for target rviz_rendering
[  3%] Built target rviz_rendering_automoc
[  6%] Linking CXX shared library librviz_rendering.so
/usr/bin/ld: /home/rpi/ros2_ws/install/opt/rviz_ogre_vendor/lib/libOgreMainStatic.a(OgreHardwareVertexBuffer.cpp.o): relocation R_ARM_THM_MOVW_ABS_NC against `__stack_chk_guard' can not be used when making a shared object; recompile with -fPIC
/home/rpi/ros2_ws/install/opt/rviz_ogre_vendor/lib/libOgreMainStatic.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
CMakeFiles/rviz_rendering.dir/build.make:583: recipe for target 'librviz_rendering.so' failed
make[2]: *** [librviz_rendering.so] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/rviz_rendering.dir/all' failed
make[1]: *** [CMakeFiles/rviz_rendering.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/rpi/ros2_ws/build/rviz_rendering/cmake__build.sh && /usr/bin/make -j1' failed in '/home/rpi/ros2_ws/build/rviz_rendering' with exit code '2'
<== Command '. /home/rpi/ros2_ws/build/rviz_rendering/cmake__build.sh && /usr/bin/make -j1' failed in '/home/rpi/ros2_ws/build/rviz_rendering' with exit code '2'
```
#### Additional information
I am currently building using:
```src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install --skip-packages ros1_bridge rviz_rendering rviz_rendering_tests rviz_common rviz2 rviz_default_plugins --make-flags -j1```

wjwwood, 2018-02-01:
This is surprising since we build it on our ARM CI:

http://ci.ros2.org/job/ci_linux-aarch64/

As for the static linking, that's no longer being used in the current ros2 branch. We'll also be doing a patch release to Ardent soon which will include updates to rviz. So I'm going to close this until you can try a more up-to-date version of rviz. If it's still a problem I can reopen this.

closed 2018-02-01 by wjwwood

Ekci, 2018-02-01:
@wjwwood: tank you for your answer. You mean this patch? [Ardent patch1 #455](https://github.com/ros2/ros2/pull/455)? There's no change that helps solve my problem. Could you guide me what should I do to compile rviz? I thought that I am on current branch.

wjwwood, 2018-02-01:
If you're following the https://github.com/ros2/ros2/wiki/Linux-Development-Setup guide then you're using the "ardent" `ros2.repos` file, which is about a month and a half old at this point.

That is the patch pr I'm talking about, but for rviz it's essentially the current `ros2` branch.

We don't build Ogre as a static library on the current `ros2` branch, so that leads me to believe you're not using the latest branches everywhere.

You can simply use this url instead and follow the above guide exactly the same to build the "latest" of everything:

```
wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
```

Ekci, 2018-02-01:
I just changed version to this in my ```ros2.repos```:
```
  ros2/rviz:
    type: git
    url: https://github.com/ros2/rviz.git
    version: d3c13b014edf11c1d7a9914db5f1d1ca33cc5e48
```
But of course, I was using ```release-latest``` instead of ```master```. I should have realized that, thank you for your time and great support :)

wjwwood, 2018-02-01:
Yeah, that's not sufficient, you'll need to use the latest of everything to get a reasonable result most of the time.

Good luck :)