ros2/rviz #207: Camera display makes unclear assumptions about camera info topic

greimela-si, 2018-02-15:
Currently the camera display assumes that the name of the camera info topic is constructed in the following way:
`<image_topic>/camera_info`

However, this is different from the original rviz which uses `image_transport::getCameraInfoTopic()` to determine the camera info topic.
This is due to the fact that the image_transport package is not availabie for ROS 2 yet.

We should restore the original behavior in time.

Myzhar, 2019-01-16:
I was going to open an issue about this same problem that is present in Crystal Clemmys.

https://answers.ros.org/question/312930/ros2-image_transport-and-rviz2-camera-something-wrong/

chapulina, 2019-07-25:
I made a PR addressing this a while back: #419. It doesn't use image transport, but fixes the behaviour so it matches ROS 1.

If that is released into Dashing, it could potentially break people who have been using the current code with some workaround.

Feedback appreciated.

Martin-Idel, 2020-03-15:
Just to let you know: I'm working with an intern on reenabling image transport throughout RViz. This should address this issue completely (hopefully for foxy)

closed 2021-05-07 by mjcarroll