ros2/rviz #208: Ardent rviz2 binaries crash at startup on Windows

greimela-si, 2018-02-19:
We experience problems with the official [ardent bugfix release binaries](https://github.com/ros2/ros2/releases/tag/release-ardent-20180209) on some Windows machines.

After sourcing the `setup.bat` and executing `rviz2` it crashes with the error
```
[ERROR] [pluginlib.ClassLoader]: Could not find package manifest 
  (neither package.xml or deprecated manifest.xml) at same directory level 
  as the plugin XML file C:\Dev\ros2/share/rviz_default_plugins/plugins_description.xml. 
  Plugins will likely not be exported properly.)
```
The `package.xml` is at the correct place next to `plugins_description.xml`.

This error only occurs on some of our Windows machines.
Rviz2 also works fine when building ros2 from source.

We could not yet identify possible causes for this behavior.
Maybe it has to something with the filesystem helper in the pluginlib?

Any ideas how we could approach this issue?

greimela-si, 2018-02-19:
I found the cause for this error.

**The pluginlib is not able to read files that do not have the archive bit set**

Additional findings:
* The error only occurs when 7zip is used to extract the archive. This is because its extraction mechanism does not set the archive bit.
* If the Windows explorer is used for extraction everything works fine.
* Another factor might be that the windows archive is compressed on a linux machine. 
7zip therefore tries to map the linux access flags to windows permissions and attributes.

The question is why the archive bit influences the readability of a file...

greimela-si, 2018-02-20:
Correction:

**The real cause seems to be the `X` file attribute.**

Reading the file fails if the `X` attribute is set and the `A` attribute is not set.
This is why I originally thought that the `A` attribute might be responsible.

greimela-si, 2018-02-20:
@wjwwood Can you add a note to the [Windows install instructions](https://github.com/ros2/ros2/wiki/Windows-Install-Binary) on how to extract the archive correctly?

wjwwood, 2018-02-20:
> @wjwwood Do you want to add a note to the Windows install instructions on how to extract the archive correctly?

👍 I'll get to that asap, but for future reference you guys can edit the wiki (anyone can actually).

greimela-si, 2018-03-01:
@wjwwood Thanks for the info, I added the respective note to the wiki.

closed 2018-03-01 by greimela-si

calvertdw, 2018-07-26:
Note this seems to work fine on bouncy.1.