ros2/rviz #224: Conflict between two yaml-cpp installations

janismac, 2018-02-27:
I'm not sure this is the right place for this issue. If not, maybe someone can point me to the right place.

I have ROS1 and ROS2 installed simultaneously. I'm using yaml-cpp in my program.

ROS1 relies on:

    Package: libyaml-cpp0.5v5:amd64
    Library: /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.5.2

ROS2/rviz relies on:

    Package: ros-ardent-rviz-yaml-cpp-vendor
    Library: /opt/ros/ardent/opt/rviz_yaml_cpp_vendor/lib/libyaml-cpp.so.0.5

My program can compile and link, but on runtime it crashes with `<my_program_path>: symbol lookup error: <my_program_path>: undefined symbol: _ZN4YAML6detail9node_data14convert_to_mapEN5boost10shared_ptrINS0_13memory_holderEEE`. This indicates that the wrong .so file is used at runtime. I temporarily fixed the problem by removing `ros-ardent-rviz-yaml-cpp-vendor`.

What would be the correct approach to distinguish between the two installations in cmake?

ROS1 version: kinetic / 1.3.1
ROS2 version: ardent / 0.4.0
Ubuntu 16.04.3 LTS

wjwwood, 2018-02-28:
> I'm not sure this is the right place for this issue. If not, maybe someone can point me to the right place.

I think this is appropriate, thanks for bringing it up.

---

Using rviz from ROS 1 should be unaffected so long as you don't run it from a shell where the `/opt/ros/ardent/setup.bash` file has been sourced. So you shouldn't need to uninstall `ros-ardent-rviz-yaml-cpp-vendor` in that case.

For running rviz2 from ROS 2, we've run into this issue where the `libyaml-cpp0.5v5` package's shared library is used rather than the one in `/opt/ros/ardent/opt/...`. This can be solved by uninstalling it, but then you have to uninstall rviz from ROS 1 which isn't great. So that's not a recommended alternative.

I believe, not 100% sure, that if you build rviz2 for ROS 2 from source when `libyaml-cpp0.5v5` is not installed, then it will always work, even if you later install `libyaml-cpp0.5v5`. So with that being said, I thought our deb `ros-ardent-rviz2` should just work, regardless. I thought we only saw this issue when building from source rviz2 and having ROS 1 already installed. I could be wrong about that though.

As for why we have a custom version distributed with rviz2 in ROS 2, the basic reason is that we're using a commit from the master branch of yaml-cpp because there's no released version (that I'm aware of) which uses C++11 instead of boost, which we wanted for our new work. It would be possible (I think) to switch our rviz2 in ROS 2 to use the libyaml-cpp from Xenial, but it present a bit of a problem, because I don't think yaml-cpp depends on boost even though it uses boost in its headers:

- no dependency: https://packages.ubuntu.com/xenial/libyaml-cpp-dev
- but used: https://github.com/jbeder/yaml-cpp/blob/release-0.5.2/include/yaml-cpp/node/detail/memory.h#L10

So then we'd need to have an explicit dependency on boost which we wanted to avoid because we use newer versions of yaml-cpp on macOS and Windows which don't need boost.

----

So as I see it we could do a few things:

- nothing, use workarounds (assuming they actually work as I described above)
  - wait for newer version of yaml-cpp in bionic (next target platform for ROS 2) and remove build of custom yaml-cpp on Linux then
- use system yaml-cpp from Xenial (avoiding distributing a custom versionon Linux)
  - either depend on boost from rviz2 or
  - consider adding boost to the rosdep rule for libyaml-cpp
- find some way to keep the current state, but avoid linking issues (rename libraries or RPATH or something else?)

In all cases we need to keep our `rviz_yaml_cpp_vendor` package until we get yaml-cpp packages on Windows and have a satisfactory solution on Linux.

wjwwood, 2018-02-28:
In all the cases, I won't have time to fix this right now, but maybe you or some of the other contributors to this repository can have a look at a solution. I will provide help as I can to make follow up releases and stuff if we find a good solution and implement it (no promises I'd get to do a new ardent release though).

Martin-Idel-SI, 2018-02-28:
We've recently experimented a bit: 
When building rviz2 from source with ROS already installed, we could fix the issue by not building the `rviz_yaml_vendor` package (comment out this line: https://github.com/ros2/rviz/blob/5394c1d560355e889c6211298f4ccc7d5c59a349/rviz_yaml_cpp_vendor/CMakeLists.txt#L55 before building).

In that case, it'll link against the system yaml-cpp used by ROS. Since that one is already built and linked against boost, this worked for us and we didn't encounter any problem so far. 

You will, however, still run into the POCO exception on shutdown: https://github.com/ros2/rviz/issues/146

clalancette, 2023-07-14:
We actually have changed this significantly since 2018, and we now use `yaml-cpp` from the system where possible.  So I believe this particular issue is fixed.  Based on that, I'm going to close this, but please feel free to reopen (or open a new ticket) if you are running into similar issues.

closed 2023-07-14 by clalancette