ros2/rviz #226: RVIZ in ROS2 cannot find freetype package

davidhuziji, 2018-03-08:
Hi, I'm cross compiling the latest ROS2 for arm platforms, following [https://github.com/ros2/ros2/tree/master/arm_crosscompilation](url)
However, I met a issue in RVIZ building.
```
The error log shows as follow.
-- Could not find freeimage library
CMake Error at /home/davhu01/project/trustzone_dds/install/share/rviz_ogre_vendor/cmake/rviz_ogre_vendor-extras.cmake:89 (find_package):
  Could not find a package configuration file provided by "freetype" with any
  of the following names:

    freetypeConfig.cmake
    freetype-config.cmake

  Add the installation prefix of "freetype" to CMAKE_PREFIX_PATH or set
  "freetype_DIR" to a directory containing one of the above files.  If
  "freetype" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
 
/home/davhu01/project/trustzone_dds/install/share/rviz_ogre_vendor/cmake/rviz_ogre_vendorConfig.cmake:30 (include)
  CMakeLists.txt:23 (find_package)
-- Configuring incomplete, errors occurred!
```
I installed the `freeimage` and `freetype6`. But it still failed.
Should I manually set the directory to freestype?

Thank you.

davidhuziji, 2018-03-08:
Some updates. I followed [https://github.com/ros2/rviz/blob/ros2/README.md](url) but it didn't work.

wjwwood, 2018-03-08:
Which OS is your host and which is your target?

Note that I didn't help write the arm cross compilation instructions, so you might want to ask the people who set that up.

mikaelarguedas, 2018-03-09:
@davidhuziji the cross compilation instructions are now hosted on the [ros2-for-arm wiki](https://github.com/ros2-for-arm/ros2/wiki/ROS2-on-arm-architecture).
As it appears that the instructions were written before the release of RViz and have not been updated since. You may want to check with @lmayencourt and @davwan01 that are maintaining the cross compilation instructions

davidhuziji, 2018-03-12:
@wjwwood @mikaelarguedas Thanks a lot. I will contact with @lmayencourt and @davwan01.
But it might be a common issue of cross compilation instructions for ROS new packages on arm platforms.
So maybe we can discuss on how to avoid such conflicts in future.

lmayencourt, 2018-03-13:
The current cross-compilation instructions hosted on the [ros2-for-arm wiki](https://github.com/ros2-for-arm/ros2/wiki/ROS2-on-arm-architecture) use `AMENT_IGNORE` to ignore a list of packages that don't compile "as it". `RViz` is one of them. 
The first idea here was to allow the cross-compilation for arm platform without any dependencies apart from the toolchain.  If you want to enable the full ROS2 stack, a deeper investigation is required.

mikaelarguedas, 2018-03-15:
as @lmayencourt pointed out, for now the cross compilation instructions are making several assumptions (some examples below):
- no Python support on the target platform
- no target system directory to use libraries already compiled for the target system

So the instructions should be updated when new packages come up and ideally reach the point where the entire stack can be cross compiled.
As you said this is not specific to this package, and opening an issue to track the status of the packages that can be cross-compiled on the [ros2-for-arm/ros2](https://github.com/ros2-for-arm/ros2) repository would be more appropriate, therefore I'm going to close this issue.

We are definitely open to review contributions to existing ROS 2 packages that would make cross-compilation easier

closed 2018-03-15 by mikaelarguedas

lmayencourt, 2018-03-22:
> We are definitely open to review contributions to existing ROS 2 packages that would make cross-compilation easier

any suggestion on how we can modify ROS2 packages for cross-compilation ?

My first thought to enable the entire stack was to provide a folder with the required libraries and set `CMAKE_SYSROOT` to it. But this solution involves others questions:
- should we use prebuild or compile from sources ?
- which version, and from where ?
- are we providing a full filesystem or only the libraries ?
- are we hosting the libraries or only the instructions get them ?
- ...