ros2/rviz #232: Bug at Installation of rviz

ChangsongYu, 2018-03-15:
Hi,

I. download rviz2 
2. ament build
3. error info:
No update step for 'ogre-master-ca665a6'
         Performing patch step for 'ogre-master-ca665a6'
         patching file CMake/Dependencies.cmake
         Performing configure step for 'ogre-master-ca665a6'
         -- ogre-master-ca665a6 configure command succeeded.  See also C:/dev/ros2-windows/rviz-ros2/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-mas
         ter-ca665a6-stamp/ogre-master-ca665a6-configure-*.log
         Performing build step for 'ogre-master-ca665a6'
         CMake Error at C:/dev/ros2-windows/rviz-ros2/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-stamp/ogre-master-ca665a6-build-Rel
         ease.cmake:16 (message):
           Command failed: 1

            'C:/Program Files/CMake/bin/cmake.exe' '--build' '.' '--config' 'Release'

           See also

             C:/dev/ros2-windows/rviz-ros2/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-stamp/ogre-master-ca665a6-build-*.log

     9>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(171,5): error MSB6006: "cmd.exe" exited with code 1. [C:\dev\ros2-windows
       \rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6.vcxproj]
     9>Done Building Project "C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6.vcxproj" (default targets) -- FAILED.
     4>Done Building Project "C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6.vcxproj.metaproj" (default targets) -- FAILED.
     2>Done Building Project "C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ALL_BUILD.vcxproj.metaproj" (default targets) -- FAILED.
     1>Done Building Project "C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\rviz_ogre_vendor.sln" (default targets) -- FAILED.

Build FAILED.

       "C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\rviz_ogre_vendor.sln" (default target) (1) ->
       "C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6.vcxproj.metaproj" (default target) (4) ->
       "C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6.vcxproj" (default target) (9) ->
       (CustomBuild target) ->
         C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(171,5): error MSB6006: "cmd.exe" exited with code 1. [C:\dev\ros2-windo
       ws\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6.vcxproj]

    0 Warning(s)
    1 Error(s)

4. open C:/dev/ros2-windows/rviz-ros2/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-stamp/ogre-master-ca665a6-build-*.log
5. I found the following error info

"C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\ALL_BUILD.vcxproj" (default target) (1) ->
"C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Bites\OgreBites.vcxproj" (default target) (2) ->
"C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj" (default target) (4) ->
(Link target) -> 
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_create_read_struct referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_set_longjmp_fn referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_create_info_struct referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_read_info referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_set_expand_gray_1_2_4_to_8 referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_set_palette_to_rgb referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_set_tRNS_to_alpha referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_set_gray_to_rgb referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_set_filler referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_set_packing referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_set_interlace_handling referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_set_strip_16 referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_read_update_info referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_read_image referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_read_end referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_destroy_read_struct referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_get_error_ptr referenced in function error_callback [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_set_read_fn referenced in function Load_SBit_Png [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  freetype.lib(sfnt.obj) : error LNK2019: unresolved external symbol png_get_io_ptr referenced in function read_data_from_FT_Stream [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
[... cut ...]
  freetype.lib(ftbzip2.obj) : error LNK2019: unresolved external symbol BZ2_bzDecompressEnd referenced in function ft_bzip2_file_reset [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]
  C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\bin\Release\OgreOverlay.dll : fatal error LNK1120: 30 unresolved externals [C:\dev\ros2-windows\rviz-ros2\build\rviz_ogre_vendor\ogre-master-ca665a6-prefix\src\ogre-master-ca665a6-build\Components\Overlay\OgreOverlay.vcxproj]

    161 Warning(s)
    31 Error(s)

ChangsongYu, 2018-03-15:
Win10 ROS2
I think, OgreOverlay need to find three libraries.
libpng.lib, zlib.lib, bzlip2.lib
+target_link_libraries(OgreOverlay C:/dev/ros2-windows/rviz2_ws/src/rviz/rviz_ogre_vendor/libpng.lib)
+target_link_libraries(OgreOverlay C:/dev/ros2-windows/rviz2_ws/src/rviz/rviz_ogre_vendor/zlib.lib)
+target_link_libraries(OgreOverlay C:/dev/ros2-windows/rviz2_ws/src/rviz/rviz_ogre_vendor/bzlip2.lib)

wjwwood, 2018-03-15:
There must be something else wrong with your environment. Ogre builds on our CI machines (and my machine locally), see: https://ci.ros2.org/view/packaging/job/packaging_windows/942/

I don't have time to try to reproduce it, but if you give more details about what you did and what might be causing this, then someone might be able to help you.

wjwwood, 2018-03-29:
I'm going to close this since we do not reproduce it in our CI, and there's not enough information to see what the difference is. I can reopen it if more information becomes available or if someone wants to pick it up.

closed 2018-03-29 by wjwwood