ros2/rviz #237: RIVZ PLUGIN WIN10 ROS2

ChangsongYu, 2018-03-28:
Hi everyone,

i want to create a rviz2 plugin and then integrate it into rviz2. However, I encountered some issues.
1. download and binary install ros2-ardent-package-windows-fastrtps-AMD64  into C:\dev\ros2
2. create rviz_test_plugins in D:\rviz_test_plugins\
3. ament build
4. error info
         rviz_test_plugins.dir\Release\flat_color_pc_transformer.obj
         rviz_test_plugins.dir\Release\intensity_pc_transformer.obj
         rviz_test_plugins.dir\Release\rgb8_pc_transformer.obj
         rviz_test_plugins.dir\Release\rgbf32_pc_transformer.obj
         rviz_test_plugins.dir\Release\xyz_pc_transformer.obj
         rviz_test_plugins.dir\Release\point_cloud_common.obj
         rviz_test_plugins.dir\Release\point_cloud_to_point_cloud2.obj
         rviz_test_plugins.dir\Release\point_cloud_selection_handler.obj
         rviz_test_plugins.dir\Release\point_cloud_display.obj
         rviz_test_plugins.dir\Release\point_cloud2_display.obj
         rviz_test_plugins.dir\Release\mocs_compilation.obj
     5>LINK : fatal error LNK1181: cannot open input file 'C:\dev\ros2_2018_03_27_fastrtps\ros2-windows\install\Lib\ament_index_cpp.lib' [D:\Repo\pj-dc_int\dc_fw\src\rviz_test_plugins\test\build\rviz_test_plugins\r
       viz_test_plugins.vcxproj]
     5>Done Building Project "D:\Repo\pj-dc_int\dc_fw\src\rviz_test_plugins\test\build\rviz_test_plugins\rviz_test_plugins.vcxproj" (default targets) -- FAILED.
     3>Done Building Project "D:\Repo\pj-dc_int\dc_fw\src\rviz_test_plugins\test\build\rviz_test_plugins\rviz_test_plugins.vcxproj.metaproj" (default targets) -- FAILED.
     2>Done Building Project "D:\Repo\pj-dc_int\dc_fw\src\rviz_test_plugins\test\build\rviz_test_plugins\ALL_BUILD.vcxproj.metaproj" (default targets) -- FAILED.
     1>Done Building Project "D:\Repo\pj-dc_int\dc_fw\src\rviz_test_plugins\test\build\rviz_test_plugins\rviz_test_plugins.sln" (default targets) -- FAILED.

Build FAILED.

       "D:\Repo\pj-dc_int\dc_fw\src\rviz_test_plugins\test\build\rviz_test_plugins\rviz_test_plugins.sln" (default target) (1) ->
       "D:\Repo\pj-dc_int\dc_fw\src\rviz_test_plugins\test\build\rviz_test_plugins\rviz_test_plugins.vcxproj.metaproj" (default target) (3) ->
       "D:\Repo\pj-dc_int\dc_fw\src\rviz_test_plugins\test\build\rviz_test_plugins\rviz_test_plugins.vcxproj" (default target) (5) ->
       (Link target) ->
         LINK : fatal error LNK1181: cannot open input file 'C:\dev\ros2_2018_03_27_fastrtps\ros2-windows\install\Lib\ament_index_cpp.lib' [D:\Repo\pj-dc_int\dc_fw\src\rviz_test_plugins\test\build\rviz_test_plugins
       \rviz_test_plugins.vcxproj]

    0 Warning(s)
    1 Error(s)

wjwwood, 2018-03-29:
I just downloaded the archive and that `ros2-windows\install\Lib\ament_index_cpp.lib` file exists. Can you confirm that it does?

There was a related issue that using some programs to extract the archive could cause problems due to permission flags, see: https://github.com/ros2/rviz/issues/208

I don't know, but I suppose this might be a related issue.

wjwwood, 2018-05-01:
Closing until more information is gathered.

closed 2018-05-01 by wjwwood