ros2/rviz #240: Current rviz2 shows error during runtime

sagniknitr, 2018-04-02:
Runnig rviz2 on terminal causes error " ViewController "Current View' unavailable.

The error message reads as : " Could not find corresponding to plugin rviz/Orbit. Make sure the plugin description XML file has the correct name of the library and that the library actually exists."

sagniknitr, 2018-04-02:
In the following file https://github.com/ros2/rviz/blob/ros2/rviz_default_plugins/plugins_description.xml ,
the description has been changed from "rviz/Orbit" to "rviz_default_plugins/Orbit". Does this cause the error ?

wjwwood, 2018-04-02:
I believe you might be using the wrong version of `pluginlib`. The one for ROS 2 will find the library correctly without your proposed pull request change (https://github.com/ros2/rviz/pull/241), see:

https://github.com/ros/pluginlib/blob/0e879d471772e162bce5d91569f07b2ac6527b52/pluginlib/include/pluginlib/class_loader_imp.hpp#L420-L430

It will even warn you if you use `lib` as the prefix.

sagniknitr, 2018-04-12:
That's strange...I built the rviz2 tool on a seperate workspace and I am sure that I am using the ros 2 branch of pluginlib to build rviz2 using ament.
Anyway, I am closing this issue as  rviz2  is working fine for me  now.

closed 2018-04-12 by sagniknitr