ros2/rviz #245: Assert triggering in nightly MacOS debug builds

nuclearsandwich, 2018-04-12:
The last two nightly debug builds on Mac have seen the following assertion in a number of rviz related tests:

```
Assertion failed: ([externalHandle isKindOfClass:[NSView class]]), function create, file /Users/osrf/jenkins-agent/workspace/nightly_osx_debug/ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/RenderSystems/GLSupport/src/OSX/OgreOSXCocoaWindow.mm, line 334.
```

Test results from unstable builds:

* https://ci.ros2.org/view/nightly/job/nightly_osx_debug/816/testReport/
* https://ci.ros2.org/view/nightly/job/nightly_osx_debug/817/testReport/

greimela-si, 2018-04-12:
Strange, the PR that introduced these changes (#186) was green on macOS:
https://ci.ros2.org/job/ci_osx/3460/consoleFull#console-section-342
The job even ran on the same slave.

Note that the new tests require a real display to run correctly, thats why they are currenly only enabled on macOS.

mikaelarguedas, 2018-04-12:
Is it possible that building them in debug mode makes a difference ?
The same test passed in the `Release` nightly, and with `None` on CI

nuclearsandwich, 2018-04-13:
Possibly also related is this failure to find a DLL on Windows debug which is causing some failures in rviz tests.

```
run_test.py: invoking following command in 'C:\J\workspace\nightly_win_deb\ws\src\ros2\rviz\rviz_common':
 - C:/J/workspace/nightly_win_deb/ws/build/rviz_common/Debug/rviz_common_ros_node_abstraction_test.exe --gtest_output=xml:C:/J/workspace/nightly_win_deb/ws/build/rviz_common/test_results/rviz_common/rviz_common_ros_node_abstraction_test.gtest.xml
-- run_test.py: return code 3221225781
```

[According to Stack Overflow](https://stackoverflow.com/questions/11432940/what-does-error-code-0xc0000135-mean-when-starting-a-net-application) the exit code 3221225781 is a failure to find a DLL.

The failures started at the same time as the Mac debug failures https://ci.ros2.org/view/nightly/job/nightly_win_deb/845/ but if they are indeed different I can move this to a separate issue.

Martin-Idel-SI, 2018-04-16:
For the Mac errors:

Yes, it turns out to be a problem with the Debug mode. The failures are correct in that they actually show that a debug build of RViz on OSX doesn't work. I created an issue in #246.

How would you like to proceed? The easiest workaround would be to skip the display tests on OSX in Debug mode. It seems unlikely that the issue can be solved fast as it seems to be a pure Ogre3D issue.

For the Windows errors:

Those are unrelated (they were introduced by the same PR, because that PR actually enabled tests on Windows. Before that, no tests ran). Since all tests in `rviz_common` fail, it seems plausible that this is due to an error in linking.
Sadly, I couldn't reproduce the errors locally. I ran a full build via
`.../ament.py build --isolated --build-tests --cmake-args -DCMAKE_BUILD_TYPE=Debug`
and then 
`.../ament.py test --isolated --skip-build --cmake-args -DCMAKE_BUILD_TYPE=Debug`.
For me, all tests passed.

nuclearsandwich, 2018-04-27:
I've closed this issue because CI is now happy as the tests are no longer being run. My understanding is that the underlying issue is being tracked in #246 which is still open.

PR #257 (2018-04-27): Disable rendering tests on Mac when building debug.
Display tests currently fail due to https://github.com/ros2/rviz/issues/246.
In order to make it easier to triage CI failures, disable these tests in that configuration pending resolution of the failure.

Temporarily resolves #245.
closed 2018-04-27 by nuclearsandwich