ros2/rviz #246: RViz does not run in Debug mode on Mac

Martin-Idel-SI, 2018-04-16:
A debug build on OSX segfaults at startup with an `assert` failing in Ogre:
```
Assertion failed: ([externalHandle isKindOfClass:[NSView class]]), function create, file /Users/osrf/jenkins-agent/workspace/nightly_osx_debug/ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/RenderSystems/GLSupport/src/OSX/OgreOSXCocoaWindow.mm, line 334.
```
This is an issue of the Ogre integration on Mac, which we thought had been solved (and actually seems to be solved in Release and ReleaseWithDebug mode), but still persists in Debug mode.

This also results in the CI-failures discussed in #245.

Martin-Idel-SI, 2018-04-26:
We tracked it to an issue with Ogre's render system (no Qt or rviz involved) and added an issue upstream. Till the issue is solved there, I guess there is not much more we can do at the moment.

wjwwood, 2018-04-26:
Thanks @Martin-Idel-SI for looking into it and opening the issue.

Martin-Idel-SI, 2018-05-08:
I had a nice conversation with the Ogre maintainer and it turns out that the issue must be on our side. I'm still confused as to why it works in Release mode, but apparently, we just hand in a correct `externalWindowHandle` later on and everything just works...

PR #258 (2018-05-16): Fix debug build on Mac
Fixes #246 

This pull request fixes the bug that caused RViz to crash on Mac when built in debug mode. (This change has also been tested on Windows and Linux.)

Because of this, the rendering tests are no longer a problem on Mac, when building debug and, therefore, we have also reverted the last commit, which disabled the tests on Mac when in debug mode.
closed 2018-05-16 by wjwwood