ros2/rviz #247: Debug mode windows unable to load dll.

nuclearsandwich, 2018-04-16:
Tests enabled in https://github.com/ros2/rviz/pull/186 are not able to run in debug configuration on Windows. They fail with the error:

```
run_test.py: invoking following command in 'C:\J\workspace\nightly_win_deb\ws\src\ros2\rviz\rviz_common':
 - C:/J/workspace/nightly_win_deb/ws/build/rviz_common/Debug/rviz_common_ros_node_abstraction_test.exe --gtest_output=xml:C:/J/workspace/nightly_win_deb/ws/build/rviz_common/test_results/rviz_common/rviz_common_ros_node_abstraction_test.gtest.xml
-- run_test.py: return code 3221225781
```

That return code is associated with a failure to find a dll [according to StackOverflow](https://stackoverflow.com/questions/11432940/what-does-error-code-0xc0000135-mean-when-starting-a-net-application).

Split from #245 which is a separate issue with debug configurations on MacOS.

Martin-Idel-SI, 2018-07-12:
Is this still an issue?

wjwwood, 2018-07-12:
I don't think so as windows debug nightly is green right now. I'll just close this and we can reopen if not.

closed 2018-07-12 by wjwwood