ros2/rviz #248: [rviz_rendering] System assimp exporting wrong library dir

mikaelarguedas, 2018-04-16:
I'm using ubuntu bionic that has assimp 4.1. As a result `rviz_assimp_vendor` doesnt build the assimp from source.

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
library_abs
    linked by target "rviz_rendering" in directory /root/ros2/ros2_ws/src/ros2/rviz/rviz_rendering
```

I narrowed it down to assimp exporting the wrong library dirs:
`ASSIMP_LIBRARY_DIRS:/usr/lib/lib/x86_64-linux-gnu`

It seems to not be handling the architecture specific dir properly and duplicate the "lib" in the provided path. So I guess we would need to convince either assimp devs or debian assimp maintainers to use GNUInstallDirs

/cc @wjwwood as maintainer
/cc @nuclearsandwich as the person wit hthe most experience with GNUInstallDirs

mikaelarguedas, 2018-04-24:
awesome, thanks @nuclearsandwich !

mikaelarguedas, 2018-04-24:
As discussed offline, we should also ping the Debian maintainer to report the issue of the current deb hoping that it gets patched before this fix makes it upstream and in a packages release

nuclearsandwich, 2018-04-24:
Debian bug (baby's first!) link: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=896835

ArkadiuszNiemiec, 2018-05-16:
Could you tell me how to workaround this problem? I have tried to build and install `assimp:master` but it did not help.

clalancette, 2023-07-14:
The original fix from Mikael was merged long ago, and current versions of assimp (5.2) don't seem to have this bug anymore.  So I'm going to close this as fixed, but feel free to reopen if you think this is still a problem.

closed 2023-07-14 by clalancette