ros2/rviz #253: Ament build error in ROS2 linux

jzp1025, 2018-04-19:
i just follow the Readme.md and try to install the rviz in ROS2 , but when the ament try to build the `rviz_default_plugins`  , it occur an error   ##:
         `By not providing "Findlaser_geometry.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "laser_geometry", but CMake did not find one.`

since the laser_geometry is in not in ros2  , how can we build succeed in this ?
thanks !

anhosi, 2018-04-19:
You need to use an up-to-date ros2 (i.e. master branch) as the laser_geometry package was only recently added to ros2.

jzp1025, 2018-04-19:
thanks a lot !
im going to reinstall the ros2

jzp1025, 2018-04-19:
@anhosi 
sorry , i have add `angles` and `laser_geometry` to the ROS2 , but when i ament build , another error occured :
`In file included from /home/tusimple/rviz2_ws/src/rviz/rviz_common/src/rviz_common/././pluginlib_factory.hpp:42:0,
                 from /home/tusimple/rviz2_ws/src/rviz/rviz_common/src/rviz_common/./display_factory.hpp:38,
                 from /home/tusimple/rviz2_ws/src/rviz/rviz_common/src/rviz_common/display_factory.cpp:31:
/home/tusimple/rviz2_ws/install_isolated/pluginlib/include/pluginlib/class_loader.hpp:45:55: fatal error: class_loader/multi_library_class_loader.hpp: No such file or directory
compilation terminated.
CMakeFiles/rviz_common.dir/build.make:158: recipe for target 'CMakeFiles/rviz_common.dir/src/rviz_common/display_factory.cpp.o' failed
make[2]: *** [CMakeFiles/rviz_common.dir/src/rviz_common/display_factory.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
`
Is there still caused for i don't reinstall the whole ROS2 ?

thanks !

jeising, 2018-04-19:
Hi @jzp1025 ,
To use the latest features, which are not part of any release, you might need to follow this installation guide:
https://github.com/ros2/ros2/wiki/Linux-Development-Setup
And use **master** instead of **latest-release** for the ros2.repos file. This way you should be able to build ros2 with rviz.
Or are you already building ros2 from source?

anhosi, 2018-04-19:
@jzp1025 I would guess so. The class_loader is part of ros2 and has seen a couple of fixes since the last release. I would recommend that you update your whole ros2 installation.

jzp1025, 2018-04-19:
@anhosi @jeising 
thanks guys !
i just add two the new two gits(`angles` and `laser_geometry`) to my ros2 , so maybe it's reason why the build failed . 
i will fully reinstall the ros2 now.

appreciate your patience ! XD

closed 2018-04-23 by jzp1025