ros2/rviz #254: Some issues about trying to send a basic shape with Marker

jzp1025, 2018-04-23:
Hi , I have built the rviz in ros2 successfully .

But when i try to send a basic shape like what the tutorials do in ros1 , i have met some problems:

1.
when i add a Grid , it shows a `Status Error : Could not transform from [base_link] to [base_link]`
(But i have set the `Fixed Frame : base_link`)

2.
i write a node to publish Marker like ros1 , but the Marker shows a Status Error:

   right : Topic : 23 messages received
   right : Array Topic : OK
   error : basic_shape/0 

   As i set my Marker like this :   

    marker = std::make_shared<visualization_msgs::msg::Marker>();

    uint32_t shape = visualization_msgs::msg::Marker::CUBE;

    marker->header.frame_id = "base_link";
    //marker.header.stamp = rclcpp::Time::now();

    marker->ns = "basic_shapes";
    marker->id = 0;

    marker->type = shape;

    marker->action = visualization_msgs::msg::Marker::ADD;

    marker->pose.position.x = 0;
    marker->pose.position.y = 0;
    marker->pose.position.z = 0;

    marker->pose.orientation.x = 0.0;
    marker->pose.orientation.y = 0.0;
    marker->pose.orientation.z = 0.0;
    marker->pose.orientation.w = 1.0;

    marker->scale.x = 1.0;
    marker->scale.y = 1.0;
    marker->scale.z = 1.0;

    marker->color.r = 0.0f;
    marker->color.g = 1.0f;
    marker->color.b = 0.0f;
    marker->color.a = 1.0;

I guess it's something about `marker->ns = "basic_shapes";` , but i dont't know how to set it
i'm wondering if there are some examples of rviz in ros2 publishing msgs that i can refer to

thanks !

Martin-Idel-SI, 2018-04-23:
Could you please publish a transform (any transform) from any frame to `base_link` and see whether everything clears up, e.g. in a terminal run `ros2 run tf2_ros static_transform_publisher -- 0 0 0 0 0 0 map base_link`?

There is a known issue with tf2 that it throws an error if it can't find a reference to a frame, even if no transform is necessary, see here: https://github.com/ros2/geometry2/issues/58

jzp1025, 2018-04-23:
> in a terminal run ros2 run tf2_ros static_transform_publisher -- 0 0 0 0 0 0 map base_link

This works ! 
the Marker is successfully shown now .
I'm trying to send image type then , thanks for your answer !

closed 2018-04-25 by jzp1025