ros2/rviz #338: Notify user when graphics support is insufficient

Martin-Idel-SI, 2018-07-16:
Current situation:

I have a Windows virtual machine and installed ROS on it. 
In a console, run `ros2 run rviz2 rviz2`, then the splash-screen appears and then nothing happens and the console returns. It seems that everything works, but it doesn't.

This is actually expected since VirtualBox support of OpenGL for Windows is only Version 1.1, this is not sufficient for our Ogre rendering systems.

Suggestion:

I'd like to have any kind of feedback telling me that the issue is missing graphics support - an error printed to the console should do.

wjwwood, 2018-07-16:
Currently I think it's actually segfaulting because Ogre just tries to call OpenGL or maybe GLX functions that may or may not be `nullptr` without checking them. I fixed one such case here: https://github.com/OGRECave/ogre/pull/507

wjwwood, 2018-07-16:
Ideally, we could find replacement functions for each missing call and get VM support back, because that was super convenient, though I didn't have the expertise or time to do all that.

Martin-Idel-SI, 2018-07-17:
For Linux VMs, it should actually work because there is enough OpenGL to support them (haven't tested in a while though). 

For Windows VMs, if we don't want to do some tricks with mesa drivers, the OpenGL version is ancient (1.1.) and not supported by Ogre. It should be far easier to get the Direct3D9 plugin working and use that instead.

wjwwood, 2018-07-17:
Unless it changed recently I was unable to test in Linux VM's on my mac, but rviz from ROS 1 still worked. That was a few months ago, but I don't think our Ogre version has changed at all.

Martin-Idel-SI, 2018-07-17:
Weird, we tested with a Linux VM (Linux host, though) and it just worked out of the box.