ros2/rviz #350: yaml-cpp vendor fails to build with newest VS2017 15.8

Martin-Idel-SI, 2018-09-05:
Problem: yaml-cpp won't build with newest Visual Studio. Error:
```<...>\yaml-cpp\exceptions.h(149): error C3646: '_NOEXCEPT': unknown override specifier (compiling source file <...>\src\parse.cpp) [<...>\yaml-cpp.vcxproj]
```

There is already a fix upstream, but it's not released, yet:
https://github.com/jbeder/yaml-cpp/commit/0f9a586ca1dc29c2ecb8dd715a315b93e3f40f79

Either we have to apply the fix ourselves or wait for the next 0.6 release and upgrade.

mikaelarguedas, 2018-09-10:
As we are already using a specific commit hash for yaml-cpp, would it be worth trying the latest master just to see if it builds successfully pending the 0.6 release?

mikaelarguedas, 2018-09-11:
thanks @Martin-Idel-SI for reporting and tracking down the right patch!

PR #353 (2018-09-11): use latest yamlcpp commit to fix compilation with vc15.8.1
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Tested locally on Visual Studio 2017 v15.8.1 and confirmed that it fixes the compilation issue

Closes #350
closed 2018-09-11 by mikaelarguedas