ros2/rviz #351: Render panel disappears on File -> Save Image

greimela-si, 2018-09-06:
When clicking `File` -> `Save Image`, a new window opens to save an image of the current scene.
However, this action clears the render panel in the main window.

[image]

Closing the new window does not restore the render panel, it stays blank.
Klicking the `Retry` button in the new window does restore the render panel.

gbalke, 2021-09-16:
It seems like this bug has resolved itself

https://user-images.githubusercontent.com/11024792/133676803-df3ecc1a-6cbd-4e72-8eb0-096cde511288.mp4

?

ahcorde, 2021-09-21:
I tried to reproduce this on rolling and galactic and I couldn't replicate it.

@greimela-si feel free to reopen the issue and add more information if the issue persist.

closed 2021-09-21 by ahcorde